noopkat/avrgirl-arduino #3: USBASP or Arduino as ISP Compatibility

chocol4tebubble5, 2015-07-27:
Is this compatibility planned, and how difficult would it be to implement?

noopkat, 2015-07-30:
Hi @chocol4tebubble5, Thanks for checking in ⭐️

Definitely on the roadmap soon, so short answer = yes. 
Note though that USBASP support is planned for a separate `avrgirl` package from this one. I would prefer to have a more generic package for USBASP, which will incidentally support Arduino as ISP out-of-the-box. The reason for this separation is because an Arduino with the 'Arduino as ISP' sketch uploaded to it presents itself as no different from a generic USBASP programmer, so we don't have to treat it as an Arduino strictly speaking.

chocol4tebubble5, 2015-07-30:
Thank you very much!

closed 2015-07-30 by chocol4tebubble5

Riz-waan, 2024-09-03:
Hi, I was just wondering if there was any update on this?