noopkat/avrgirl-arduino #4: Write tests

noopkat, 2015-09-10:
Please see [avrgirl-stk500v2 tests](https://github.com/noopkat/avrgirl-stk500v2/tree/master/tests) and the [travis file](https://github.com/noopkat/avrgirl-stk500v2/blob/master/.travis.yml) for a good starting point on this. Serialport will need to be stubbed, an example is in the repo linked.

mcdonnelldean, 2015-09-11:
@noopkat Is this something that can be done sans hardware? I wouldn't mind taking a crack at tests over the weekend if that's the case. Although if they are already being done that's cool too, either way I have free cycles over the weekend that I would like to use on some grunt work to get a feel for the libs :dancer:

noopkat, 2015-09-11:
@mcdonnelldean thanks for offering to help! No one is currently working on tests, I lodged this issue just yesterday. This part of AVRGirl will be particularly time consuming to write tests for, so feel free to only pick one part of it if you like.

Yeah this should probably be done sans hardware in general (so that we know TravisCI will definitely be able to run the tests), similar to the repo I linked. And it is possible to write all of the tests without ever needing an Arduino. 

I think that we shouldn't need to write tests for every single piece of hardware, but more that the properties being passed around for each set of hardware is correct. Not sure if that makes sense!

Without hardware it can be weird to ramp up on the context of each method, but in the case of this library I think it's easy to understand the hardware workflow as the Arduino platform is pretty familiar to you right? You could probably stub out the serialport and then turn debug mode on with some additional console logs to get an idea of what happens in which order. If you have an Uno at the bare minimum it could be useful just to run it a few times to see how it all works, but not necessary. The Uno makes use of the stk500v1 protocol if that helps you get started.

Here's how I see the tests in my head:
+ Stub out node serialport with [virtual-serialport](https://www.npmjs.com/package/virtual-serialport), making use of [proxyquire](https://github.com/thlorenz/proxyquire) most likely
+ Stub out any additional methods, responses and properties needed (virtual-serialport is not a complete mock).
+ Create a series of tests for each protocol workflow by setting up a virtual device with virtual-serialport:
  + stk500v1
  + stk500v2
  + avr109
+ Use sinon.js to spy on the upload flow to see that the correct methods are being called, with correct args and call counts, etc.

There is some refactoring I'd like to do on this library, as it's not super readable/easy to follow and everything is heaped into one big library file. I want to remove some callback stacks, and split each protocol out properly to ease all of the conditionals out. Tests would help out with the refactor process SO MUCH :heart_eyes: 

I have turned on this repo in my TravisCI account, if you don't already have an account I'd recommend signing up and adding your fork to it :dancer: 

I'm here to help if you decide to tackle this (or for anyone else), it can be really confusing to know where to start. Let me know your thoughts!

mcdonnelldean, 2015-09-12:
@noopkat Awesome reply!

So I think it makes the most sense for me to start with running against my Arduino to figure out what is going on. I know a little about hardware but I'm still very much a novice, the hope is by helping out on tests I'll get a better understanding of what's going better.

Once I get through that I'll move to stubbing out and trying the easier tests. What I will say is don't let me hold you up. I'm unsure how long it will take me to ramp up, but hopefully not too long :smile: 

I also have @mcollina to help me out too. I think having two hardware superstars to keep me on the right path should make this simpler onboarding :dancer:

Did you get a chance to do a shopping list yet? It would be awesome to get hold of some more chips and a programmer.

Thanks a mill for such an awesome response!

mcdonnelldean, 2015-09-12:
@noopkat Have you any preference on testing / assertion suites? I'm partial to [Tape][] / [Faucet][], which is a TAP based suite and a pretty print reporter. Having said that I've used pretty much most of the more popular one and will defer to whichever you feel would better suit. 

[Tape]: https://www.npmjs.com/package/tape
[Faucet]: https://www.npmjs.com/package/faucet

mcdonnelldean, 2015-09-12:
Oh, I just found https://github.com/noopkat/avrgirl-stk500v2/blob/master/tests/avrgirl-stk500v2.spec.js so I'll follow the conventions there.

noopkat, 2015-09-12:
@mcdonnelldean awesome! This is an example of how the tests are set up in the package json: https://github.com/noopkat/avrgirl-stk500v2/blob/master/package.json#L7 I normally pipe into tap-spec cos it's easier to read for me :smile: I love tape, sounds like we're on the same page!

Shopping list - I'm on it. It'll be mostly what avrgirl currently supports. Should have that done tonight.
Tonight I'm also trying to get most of the grunt work done to translate all atmega and attiny chips over to the avrgirl spec json format. From there it'll be much more clear what needs physical hardware testing. 

Thank you so much again!

noopkat, 2015-09-12:
@mcdonnelldean also this here is how I have previously set up and stubbed the virtual-serialport instance: https://github.com/noopkat/avrgirl-stk500v2/blob/master/tests/serialport-comms.spec.js#L6-L33

There were a few missing methods, so as you can see I just used threw them in a bit unceremoniously but it worked. I should open a PR on the virtual-serialport lib.

mcollina, 2015-09-13:
I'm really looking forward to the shopping list :).

closed 2015-11-29 by noopkat