noopkat/avrgirl-arduino #6: Board support: Pinoccio Scouts

noopkat, 2015-09-11:
[Pinoccio Scouts](https://pinocc.io/) use the stk500v2 protocol, which is supplied in avrgirl-arduino already. An addition to `boards.js` should be all that is required, with some testing with a real device. I no longer have a Scout to test with. Adding a precompiled version of both Standard Firmata and Blink sketches would also very much be appreciated.

makenai, 2015-10-04:
I've been working on this,  but getting bad results. For example, using avrgirl, when I try to verify the signature, I get `0x30, 0x10, 0x00` instead of the expected signature `0x1e, 0xa8, 0x02` which is in the shipped bootloader. When I try to use the example that ships with stk500v2, I get completely random results instead.

Have you verified that the mega works?

makenai, 2015-10-04:
Ignore previous comment. This is tentatively working and in for PR soon. :)

noopkat, 2015-10-04:
thank you! What was the issue, if you don't mind sharing?

makenai, 2015-10-04:
It was a silly mistake - at some point between harvesting the hexfiles and verifying that the flash worked with the Arduino IDE, I turned off power to the pinoccio. It was still responding as a USB device and connecting and had lights when it was connected, but the processor was not powering on and so I guess I was just picking up random noise!

noopkat, 2015-10-04:
Ah, thanks for explaining! I recognised your 'signature' as an echo of the get sig command bytes being returned back, so I figured something might not have been making it through.

Thanks for working on this issue!

noopkat, 2015-10-04:
I'm sure you're aware of this, but trying to save you from work just in case - the Pinoccios have their own ecosystem, so the standard blink and firmata sketches might not be useful to include, or not worth the time making them compatible if there's extra work needed. 

If there's an obvious 'hello' world hex you can compile that does not rely on individual config settings etc, then feel free to include it in junk/hex/pinoccio. Otherwise, don't worry about including any hex files at all.

makenai, 2015-10-04:
Yup, I'm aware of the differences, however I'd like to include both varieties for sake of completeness. pinnocio-io uses Bootstrap.cpp.hex instead of Firmata, and there is a special LEDBlink.cpp.hex that works with the pinoccio torch. It is also possible to use Blink.cpp.hex, and this morning I left off forking Firmata to add some flags to make it work. I should be able to test that this evening.

It's mostly a vanity thing for myself since (as you reminded me) I now only have the single Pinoccio I originally crowdfunded and can't use pinoccio-io. I can leave whichever hexfiles you don't like out. The Pinoccio-specific ones are about 400k each since they include the entire Bitlash stack. :/

noopkat, 2015-10-04:
:+1: good to know, thank you!

PR #16 (2015-10-06): Add support for Pinoccio
Closes #6 

Tested on a regular Pinoccio Scout, but should work the same on a Lead Scout.  I also added a warning message about specifying an invalid board type. The hexfiles I included have some special gotchas, so I added a README there too.
closed 2015-10-06 by noopkat