noopkat/avrgirl-arduino #8: Support Pro Mini

noopkat, 2015-10-02:
This is a priority issue, in order to support @ajfisher's great work on his [nodebots-interchange](https://github.com/ajfisher/nodebots-interchange) project.

A few things to consider here:

+ No reliable pid or vid, as the FTDI adaptor will most likely report itself instead. Brute force 'sign-on' attempts may be needed to attempt to get an expected answer back from one of the active ports.

+ Board reset method may differ depending on which FTDI adaptor is used (breakout board vs cable). Specifically this is pulling either the DTS or RTS low. More research to confirm this is needed.

+ Naming conventions so far has me leaning towards `pro-mini` as the string based identifier.

+ STK500v1 is the protocol to use for this board.

ajfisher, 2015-10-02:
So I reckon we can make some assumptions on this because if you're using a
pro mini you're probably a bit more at the experienced end of the arduino
/avr world.

Do these assumptions make it a bit easier / straightforward?

* if you can't detect a port then bail, don't brute force it, thus
expectation is that I supply a port to connect to (as it could be pretty
varied)

* reset will be mediated by the user in all cases (even under arduino ide
you have to do this a lot to trigger an upload)

`pro-mini` works for me, I'll align Interchange to this as well for
consistency.

noopkat, 2015-10-02:
Thanks for this input - this makes a lot of sense. I guess the 'pro' in the board name should be indicative :grin: Let's go with your suggestions and get this board supported.

I added the Pro Mini to the boards list on a new test branch, if you want to jump ahead and test it before my pro mini arrives in the mail tomorrow, feel free to point avrgirl-arduino in your package json at `git://github.com/noopkat/avrgirl-arduino#pro-mini` and then run `npm install` again to have it swap out the published version. If there's an error, or it does't work let me know and I'll fix it. 

Otherwise, I'll run some physical tests on Saturday. If it works, I'll add extra documentation to the readme on this and then publish a new version to npm.

Thanks again :raised_hands:

noopkat, 2015-10-03:
Pro Mini arrived, FDTI breakout did not. Uno husk stand-in to the rescue!
Seems to be working, @ajfisher. Been pretty solid without a manual board reset too. I'll merge this and release it. 

[image]

noopkat, 2015-10-03:
released in 1.4.4.

closed 2015-10-03 by noopkat