noopkat/avrgirl-arduino #12: Human tester needed: Little Bits Arduino module

noopkat, 2015-10-03:
I do not own any sets from Little Bits. I added support a while ago but cannot test it at the moment.

If anyone has this module, could they please compile some form of 'hello world' script for it (such as a blink sketch) and run the following code to see if it works as expected?

```javascript
var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'little-bits',
  debug: true
});

avrgirl.flash('Blink.cpp.hex', function (error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});

```

Thanks a million :yellow_heart:

nodebotanist, 2015-10-19:
I will try to get a hold of one-- my dad for sure owns one, I'll see if he'll let me borrow it. If I can, I'll bring it to NYC later this week :D.

noopkat, 2015-10-20:
you are the best, thank you!

noopkat, 2015-10-26:
confirmed working, thanks for testing this @nodebotanist :yellow_heart:

closed 2015-10-26 by noopkat