noopkat/avrgirl-arduino #29: Consider replacing leo-reset with the DTR reset used for STK500 boards

ArekSredzki, 2015-10-18:


noopkat, 2015-10-18:
I just ran a quick test with this, and lowering DTR closes the port fully (so that we can reconnect at the correct baud rate for programming). Thanks for the suggestion and research on this :grin: 

Now if we could only get `serialport.close` to do the same :wink: I remember tearing my hair out at the time when trying to get a reliable reset for all of the leo boards, with only [this reset documentation](https://www.arduino.cc/en/Main/ArduinoBoardLeonardo) to follow from.

I'll get the leo reset patched up this week. Thanks!

ArekSredzki, 2015-10-18:
@noopkat No worries, I know the feeling. This Auto-reset seems to work on all arduino boards. In my Arduino wrapper I simply reset the board on serial port connect. I'll separate out the module from my project and publish it on npm soon so that you can take a look at what works for me.

It's important to note that auto reset on DTR can be disabled in a sketch (though rare for people to do). And it also won't happen if the program crashes from a bug, which sucks if you want to flash the arduino with an update to fix that bug automatically. I believe that you can only fix that with a pulse to the reset pin (untested), or by depowering the device.

noopkat, 2015-10-27:
published in 1.4.12

closed 2015-10-27 by noopkat

ArekSredzki, 2015-10-28:
@noopkat I've extracted that module I mentioned :)

https://www.npmjs.com/package/arduino-interface

noopkat, 2015-10-28:
@ArekSredzki thank you, this is very awesome!

ArekSredzki, 2015-10-28:
:smile: