noopkat/avrgirl-arduino #30: On Arduboy flashing fails with "Cannot open" or mid-flash inconsistently

flaki, 2015-10-22:
[Arduboy]() is a tiny gameboy-like device, based around the [Arduino Leonardo](http://community.arduboy.com/t/reference-page-for-development/76/6). My [dev kit](https://www.tindie.com/products/bateske/arduboy/) has just arrived. It runs and flashes fine with the Arduino IDE, but with `avrgirl-arduino` I just couldn't get it to flash the generated `hex`.

It usually doesn't work, simply fails to access the device:
```
MacBooks-MacBook-Air:avrgirl-arduino flaki$ node bin/cli.js flash -f ~/Data/ArduBreakout/examples/ArduBreakout/ArduBreakout.cpp.leonardo.hex -a leonardo -v
found leonardo on port /dev/cu.usbmodem1411
resetting board...
Failed to connect: # 1
reset complete.
[Error: Cannot open /dev/cu.usbmodem1411]
MacBooks-MacBook-Air:avrgirl-arduino flaki$ 
```
*(fwiw, the board does reset on every run of the above command, consistently).*

Sometimes it then seems to work, but amidst flashing an error is thrown, and the device flash is left in a garbled state (that is, flashing overwrote some of the installed data, so the device wouldn't "start"). Reflashing with a separate tool (like Arduino IDE) "fixed" the state of the flash, so at least it doesn't brick the device or anything like that:
```
MacBooks-MacBook-Air:avrgirl-arduino flaki$ node bin/cli.js flash -f ~/Data/ArduBreakout/examples/ArduBreakout/ArduBreakout.cpp.leonardo.hex -a leonardo -v
found leonardo on port /dev/cu.usbmodem1411
resetting board...
Failed to connect: # 1
Failed to connect: # 2
reset complete.
connected
flashing, please wait...
/Users/flaki/Data/avrgirl-arduino/node_modules/chip.avr.avr109/chip.avr.avr109.js:156
        throw err;
              ^
Error: TODO: Start Linear Address Record
    at /Users/flaki/Data/avrgirl-arduino/node_modules/chip.avr.avr109/node_modules/intelhex/index.js:57:28
    at Array.forEach (native)
    at /Users/flaki/Data/avrgirl-arduino/node_modules/chip.avr.avr109/node_modules/intelhex/index.js:11:11
    at process._tickCallback (node.js:355:11)
MacBooks-MacBook-Air:avrgirl-arduino flaki$ 
```

flaki, 2015-10-23:
Hmm! Apparently the error is generated in [`chip.avr.avr109`](https://github.com/tmpvar/chip.avr.avr109/blob/master/chip.avr.avr109.js#L156), but originates in [`intelhex`](https://github.com/tmpvar/intelhex/blob/master/index.js#L57).

According to the ARM knowledge base
> The Start Linear Address Record can appear anywhere in hex file. **In most cases this record can be ignored** because it does not contain information which is needed to program flash memory"

Commenting out the line does indeed fix it and `avrgirl-arduino` successfully flashes the hex.

The intermittent reset problem is still there, though. :confused:

flaki, 2015-10-23:
Oh dear I'm such a doofus! Thanks for the quick acting, but as it turns out I had a typo in the code! With the typo fixed, it should be working.

Also, re: `intel-hex`, `avr-arduino` does [use](https://github.com/noopkat/avrgirl-arduino/blob/master/package.json#L34) `intel-hex` over `intelhex`, but pulls in [/tmpvar/chip.avr.avr109](/tmpvar/chip.avr.avr109), which still uses `intelhex`. I'll open an issue and try reworking it to use `intel-hex` instead. Also `intelhex` is pinned on 0.0.1 version, which is why I manually had to change that for it to pull in the new release (that's when I realized I made a typo...).

noopkat, 2015-10-25:
Thanks @flaki ! Excited that you gave the Arduboy a go with avrgirl, and also I am jealous you have the dev kit!

I'm refactoring the leonardo reset script this week, and it's possible that the work will fix this intermittent issue. Are you up for testing this in a pre-flight branch for me once I'm done? A colleague might have also backed the Arduboy, I'll check in to see if I can get my hands on one so I can test it out with you.

flaki, 2015-10-25:
Hey @noopkat! Absolutely, just let me know when you need anything tested!

noopkat, 2015-10-25:
@flaki thanks so much! I just pushed up a POC branch (needs cleaning up but the functionality is there) called `leo-reset-dtr` https://github.com/noopkat/avrgirl-arduino/tree/leo-reset-dtr

`npm install && node bin/cli.js flash -a leonardo -f <file> -v` within the cloned repo (as it looks like you've been doing) should be a good test.

You may need to still have the intelhex patch in place, as looks like the bugfix hasn't been merged and published yet.

Out of curiosity - are you working with OSX Mavericks, Yosemite, or El Capitan? This'll help me look for similar kinks with my other leonardo boards I have here.

flaki, 2015-10-25:
@noopkat most welcome! BTW I'm on El Capitan. ;)

Also I've just checked out the experimental branch (and erm.. checked out the experimental branch :relaxed: ). Good news: this **works like a charm**! Five out of five times it flashes the board without any issues!
Excellent work, thank you!

If you need any more testing done with the Arduboy, feel free to let me know anytime.
 
```
MacBooks-Air:avrgirl-arduino flaki$ node bin/cli.js flash -a leonardo -f ../arduboy-rund-ino/rund/rund.ino.leonardo.hex -v
found leonardo on port /dev/cu.usbmodem1411
resetting board...
reset complete.
connected
flashing, please wait...
flash complete.
```

closed 2015-10-25 by flaki

noopkat, 2015-10-25:
thanks so much for your help @flaki, really appreciate it. I'll get this change published soon.
I'll also document in the README that although it's not directly supported, the Arduboy will work if you use `leonardo` as the board type.