noopkat/avrgirl-arduino #43: Serial read/write - 750ms delay necessary?

JonLim, 2015-12-11:
Somewhat an out-of-scope issue here, so I apologize if this is the wrong place for this, but I extended `avrgirl-arduino` to be able to send a message and wait for data back over serial, and it works beautifully. I hijack the automatic port finder and board validation to have a serialport instance ready to go before I send a message. Currently, I flashed one of our testing Arduino boards with an echo-echo Sketch, where it just spits back the data it reads over serial.

That said, I see an odd behaviour: I use a `serialPort.on('data',` function to wait for communications from the Arduino, but I have to wait approximately 750ms (minimum) after opening a connection before I can send a String over serial, and I can't do anything immediately after or the data coming back will be garbled.

Is this something you've noticed at all? Happy to provide code if you're interested in what I'm having it do, and I know it's outside of scope, just wondering if this is any behaviour you've seen before.

Thanks!

noopkat, 2015-12-12:
oh! This is really neat! Are you using the Mega still for this? If not, which board? I'm using three different protocols under the hood, so I'd need to start with the correct one to answer your questions in depth :smile:

JonLim, 2015-12-13:
@noopkat Still using the mega, and I'll post my code so you can see just what nonsense I'm getting up to.

```
Avrgirl.prototype.sendMessage = function(data, callback) {
  var _this = this;

  // validate board properties first
  _this._validateBoard(function(error) {
    if (error) { return callback(error); }

    // set up serialport connection
    _this.connection._init(function(error) {
      if (error) { return callback(error); }

      _this.connection.serialPort.open(function (error) {
        if ( error ) {
          console.log('failed to open: '+error);
        } else {
          console.log('open');

          _this.connection.serialPort.on('data', function(data) {
            console.log('data received: ' + data);
          });

          // Wait 750ms and then send data, in order to get an echo back
          setTimeout(function() { _this.connection.serialPort.write(data); }, 750);

        }
      });

    });
  });

};
```

Essentially a carbon copy of the initialization for your `.upload()` function, if I remember correctly, and I just take over `serialPort` for the opened connection from there. As for the `setTimeout()` function that writes the data, anything lower than 750 (have tested 700) and it receives nothing back.

Appreciate the help, Suz!

noopkat, 2015-12-20:
@JonLim ok! So I had a hunch about what might be causing the delay, but I had to actually play with an Arduino to prove it. Please forgive the radio silence!

My theory is that this delay is isolated to Arduino boards. Essentially, when the board's serialport is opened, the board is automatically reset and the software takes some time to reboot. This is by design, and it's due to the wiring of the board itself I believe. The delay therefore is tied to how long it takes for the board to reboot and start running your sketch again.

You can test this out with two different methods: 

1. Using the Arduino IDE, and opening up the Serial Monitor. If you have any startup led blinks in your setup script (which can be a good test), you'll see them blink again, cos the board reboots as the connection opens for the Serial Monitor. I remember reading a while back that opening the Serial Monitor resets the Arduino no matter what. You'll see a delay there too before you get reads coming in.

2. Put a few led blinks into your setup script and connect using [node-serialport](https://github.com/voodootikigod/node-serialport). You will probably see the same behaviour. I think on average the delay can be anything between 600ms to 1000ms (yikes!).

SO. There are a few sneaky ways to fix this, which is essentially disabling the auto DTR style reset. Here is a link: http://playground.arduino.cc/Main/DisablingAutoResetOnSerialConnection

The catch? Disabling the DTR pin will also stop the Arduino from being able to be reset into bootloader mode, which is necessary for AVRGirl Arduino to work when you're flashing new code to the board. So flashing the board will no longer be possible unless you manually press the reset button each time. Gah! Not sure if you wanted to use the flashing utility of AVRGirl Arduino or not, so if you don't need it, then go ahead and disable the DTR with a capacitor :smile: 

Wrapping the first write in a `setTimeout` like you explained looks to be the only way to preserve all the functionality you'd like. 

Hope this helps clear things up a little bit!

JonLim, 2015-12-22:
Hiyo @noopkat, apologies on the delay in responding.

Thank you so much for looking at this in-depth! I had completely forgotten about the reboot process when opening a serial port (like opening the Serial Monitor in Arduino IDE) which might cause the delay. I don't think we'll be disabling the DTR pin, so I think we will be sticking with the `setTimeout` method.

Thanks again, can't wait to keep plugging away with `avrgirl-arduino`!

Cheers.

closed 2015-12-22 by JonLim

noopkat, 2015-12-22:
:sparkles: np, let me know if you run into anything else!