noopkat/avrgirl-arduino #50: Consider wording of project description

yaakov-h, 2016-01-18:
The project description confuses me.

> An avrgirl-wrapper library for flashing Arduino microcontroller boards with NodeJS.

To me this sounds like a library that somehow puts NodeJS on Arduino boards, which leaves me at first amazed, then severely disappointed.

A clearer description could be:

> An avrgirl-wrapper NodeJS library for flashing Arduino microcontroller boards.

noopkat, 2016-01-18:
Ah, yes, I can see why the grammar of this tagline would be very misleading. Thanks for catching this, and apologies for the rollercoaster of feelings there! Running NodeJS on an Arduino board *would* be amazing. It's even quite difficult on a Yun.

I have fixed this in newly published version 1.6.4, commit is https://github.com/noopkat/avrgirl-arduino/commit/c73f7019a1a97cd3195d3e1f1f90cfd550c0e129 Hopefully that makes things crystal clear.

Thank you again.

closed 2016-01-18 by noopkat

yaakov-h, 2016-01-18:
Much clearer, thank you.