noopkat/avrgirl-arduino #84: Flashing arduino pro micro fails

slacker87, 2016-08-26:
Flashing a basic hex file onto an arduino pro micro fails with "Error: could not reconnect after resetting board."
debug shows it finding the micro, and on the correct port.
debug then shows it resetting the board, and my pro micro does in fact reboot
then.... error, help!

noopkat, 2016-08-26:
Hi! Thanks for opening this issue. I noticed this recently when I was testing with a Micro. If you run the code again immediately after it fails the first time, it does work for me. This is obviously a bug, so this gives me a hint on where to took to start tracking it down with a fix. Thanks and I'll leave this open until I find a solution, which hopefully should be quite soon 😁

slacker87, 2016-08-26:
Yay a workaround! TY I was worried I would have to mess with serial npm packages again :fearful:

noopkat, 2016-08-26:
My aim is for folks to never have to do this stuff manually themselves starting from scratch 😄 

If you can run the `flash` method again upon the first error as a workaround for now, I'll update you when the bug is fixed. I appreciate your kindness and am happy you're getting use out of avrgirl-arduino 🙆

slacker87, 2016-08-26:
No dice in my testing....

it finds the micro, resets, and fails, but when rerun immediately after it either:

1. resets the board again
2. gives "Error: opening COM12: File not found();"
3. gives "Error: no Arduino 'micro' found"
4. gives "Error: Opening COM12: Access denied(...)"

the error depends on when in the process i have it re-run the flash command, help please :(

noopkat, 2016-08-26:
gotcha. Yeah I'd say the long bootloader time on the Micro will not allow an immediate reattempt. I'll look into this further asap, I'm sure it's solvable as it's only a recently introduced bug. More soon!

slacker87, 2016-08-26:
Ok so more testing.... I got it to flash, using the 'leonardo' board and manually forcing the correct com port...

```
var avrgirl = new Avrgirl({
				  board: 'leonardo',
  				port: 'COM12',
				  // turn on debug mode!
				  debug: true
				});

```
Hopefully this makes the fix easier knowing it reboots/flashes fine using whatever you set for leonardo? Maybe just use the same for micro?

noopkat, 2016-08-26:
oh, odd! I have a feeling I might know what it going on. This has been super helpful from you, thank you!! Lemme do a bunch of tests on windows and report back.

noopkat, 2016-08-26:
Couple of questions, as I cannot replicate this on my mac or my windows box:

1. is your Arduino Micro from Arduino, or is it a clone? It should look like [one of these](https://www.arduino.cc/en/Main/ArduinoBoardMicro)

2. Does the script work at all if you specify the port, but leave it as `micro` being the board type?

3. If you answer is 'no' to the second question, could you please run the following with your Arduino plugged in, and paste the results in this thread?

```javascript
var Avrgirl = require('avrgirl-arduino');
var avrgirl = new Avrgirl({
  board: 'micro',
  debug: true
});

var hex = 'path/to/your/hex/file.hex';

avrgirl.list(function(err, ports) {
  console.log('1st list', ports);
});

avrgirl.flash(hex, function(error) {
console.log(error);
 avrgirl.list(function(e, ports) {
    console.log('2nd list', ports);
  });
});
```

noopkat, 2016-08-27:
Me again - I just published what I think should be a fix for this bug, if you `npm install avrgirl-arduino@latest` could you please let me know if you see it fixed on your end too?

Thanks!

slacker87, 2016-08-27:
No dice, to answer your questions:

1. this is a clone board, arduino pro micro
2. same error
3. 
`Navigated to file:///C:/Users/Mike/Documents/GitHub/CoreIR-Uplink/source/index.html
C:\Users\Mike\Documents\GitHub\CoreIR-Uplink\source\node_modules\avrgirl-arduino\lib\connection.js:25 found micro on port COM11
C:\Users\Mike\Documents\GitHub\CoreIR-Uplink\source\node_modules\avrgirl-arduino\lib\avr109.js:114 resetting board...
index.html:97 1st list [Object, Object]0: Object_standardPid: "0x8037"comName: "COM11"locationId: undefinedmanufacturer: "Arduino LLC (www.arduino.cc)"pnpId: "USB\VID_2341&PID_8037&MI_00\7&179A8F7E&0&0000"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Object__defineGetter__: __defineGetter__()__defineSetter__: __defineSetter__()__lookupGetter__: __lookupGetter__()__lookupSetter__: __lookupSetter__()constructor: Object()arguments: nullassign: assign()caller: nullcreate: create()defineProperties: defineProperties()defineProperty: defineProperty()freeze: freeze()getOwnPropertyDescriptor: getOwnPropertyDescriptor()getOwnPropertyNames: getOwnPropertyNames()getOwnPropertySymbols: getOwnPropertySymbols()getPrototypeOf: getPrototypeOf()is: is()isExtensible: isExtensible()isFrozen: isFrozen()isSealed: isSealed()keys: keys()length: 1name: "Object"preventExtensions: preventExtensions()prototype: Objectseal: seal()setPrototypeOf: setPrototypeOf()__proto__: ()<function scope>hasOwnProperty: hasOwnProperty()arguments: nullcaller: nulllength: 1name: "hasOwnProperty"__proto__: ()<function scope>isPrototypeOf: isPrototypeOf()arguments: nullcaller: nulllength: 1name: "isPrototypeOf"__proto__: ()<function scope>propertyIsEnumerable: propertyIsEnumerable()arguments: nullcaller: nulllength: 1name: "propertyIsEnumerable"__proto__: ()<function scope>toLocaleString: toLocaleString()arguments: nullcaller: nulllength: 0name: "toLocaleString"__proto__: ()<function scope>toString: toString()arguments: nullcaller: nulllength: 0name: "toString"__proto__: ()<function scope>valueOf: valueOf()arguments: nullcaller: nulllength: 0name: "valueOf"__proto__: ()<function scope>get __proto__: __proto__()arguments: nullcaller: nulllength: 0name: "get __proto__"__proto__: ()<function scope>set __proto__: __proto__()arguments: nullcaller: nulllength: 1name: "set __proto__"__proto__: ()<function scope>1: Object_standardPid: ""comName: "COM1"locationId: undefinedmanufacturer: "(Standard port types)"pnpId: "ACPI\PNP0501\1"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Objectlength: 2__proto__: Array[0]
index.html:74 [Object, Object]0: Object_standardPid: "0x8037"comName: "COM11"locationId: undefinedmanufacturer: "Arduino LLC (www.arduino.cc)"pnpId: "USB\VID_2341&PID_8037&MI_00\7&179A8F7E&0&0000"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Object1: Object_standardPid: ""comName: "COM1"locationId: undefinedmanufacturer: "(Standard port types)"pnpId: "ACPI\PNP0501\1"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Objectlength: 2__proto__: Array[0]
index.html:101 Error: could not reconnect after resetting board.(…)

2nd list [Object, Object]0: Object_standardPid: ""comName: "COM1"locationId: undefinedmanufacturer: "(Standard port types)"pnpId: "ACPI\PNP0501\1"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Object1: Object_standardPid: "0x0036"comName: "COM10"locationId: undefinedmanufacturer: "Arduino LLC (www.arduino.cc)"pnpId: "USB\VID_2341&PID_0036\6&32FB597E&0&1"productId: undefinedserialNumber: undefinedvendorId: undefined__proto__: Objectlength: 2__proto__: Array[0]concat: concat()constructor: Array()copyWithin: copyWithin()entries: entries()every: every()fill: fill()filter: filter()find: find()findIndex: findIndex()forEach: forEach()includes: includes()indexOf: indexOf()join: join()keys: keys()lastIndexOf: lastIndexOf()length: 0map: map()pop: pop()push: push()reduce: reduce()reduceRight: reduceRight()reverse: reverse()shift: shift()slice: slice()some: some()sort: sort()splice: splice()toLocaleString: toLocaleString()toString: toString()unshift: unshift()Symbol(Symbol.iterator): values()Symbol(Symbol.unscopables): Object__proto__: Object`

noopkat, 2016-08-27:
out of curiosity - are you using this in a browser, Electron, or Chrome app perhaps? Trying to narrow this down. I've been testing in my node terminal, so I fear I've been testing this incorrectly the whole time

slacker87, 2016-08-27:
https://github.com/slacker87/CoreIR-Uplink

Electron, like I said, works great with leonardo selected! Who knows why :p

noopkat, 2016-08-27:
it seems that the problem is that the clone reports a non standard (ie. not a genuine Arduino) product id when in bootloader mode. Clones can be really tricky, and I can't always guarantee support for them out of the box as they widely vary with how they present as a device on USB.

I added that product id to the micro identification object and published it - could you possible retry again? Thanks!

slacker87, 2016-08-27:
Success on three different type of 32u4 clone boards! 🎉 

Now when I code up and share my project I can safely know they'll work first try with avrgirl 🎂 ty

closed 2016-08-27 by slacker87

noopkat, 2016-08-27:
🙌  🙌  🙌  🙌 

great to have another clone out in the wild tracked down and now compatible. Thanks for you patience! 🎉

ts-ycc, 2017-02-08:
I'm getting the same "File  not found" error, same as slacker87's first comment. Checked my product ID (productID: '8037') and it seemed okay although it is a clone micro.  I then used the code you supplied on August 26 and here is the result:

1st list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\8&306F1381&0&0000',
    locationId: '0000.001a.0000.001.001.004.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]
found micro on port COM7
resetting board...
Error: Opening COM7: File not found
    at Error (native)
2nd list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\8&306F1381&0&0000',
    locationId: '0000.001a.0000.001.001.004.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]

Help Please and Thank you!

noopkat, 2017-02-09:
hmmm might be a timing issue 🤔 I'll try this on Windows when I get a chance to see if I can replicate.

sandeepmistry, 2017-02-09:
Could you check the PID in bootloader mode? Just double tap the reset button and run the list command again. It might be missing for the PID list.

ts-ycc, 2017-02-09:
I reset the board twice ( there is no reset button so I jumped the GND and RST twice) and the results are the same:

No port auto sniff
1st list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\8&306F1381&0&0000',
    locationId: '0000.001a.0000.001.001.004.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]
found micro on port COM7
resetting board...

After that it just hangs. :(

slacker87, 2017-02-09:
try 'leonardo' instead of micro for board type

ts-ycc, 2017-02-09:
No luck with 'leonardo':

No port auto sniff
1st list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\8&306F1381&0&0000',
    locationId: '0000.001a.0000.001.001.004.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]
Error: no Arduino 'leonardo' found.
    at C:\Users\ts-ycc\node_modules\avrgirl-arduino\lib\connection.js:33:
25
    at C:\Users\ts-ycc\node_modules\avrgirl-arduino\lib\connection.js:74:
12
    at C:\Users\ts-ycc\node_modules\avrgirl-arduino\lib\connection.js:212
:12
2nd list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\8&306F1381&0&0000',
    locationId: '0000.001a.0000.001.001.004.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]

noopkat, 2017-02-09:
Thanks for posting all of the debug info 😄 
I'm currently traveling until later this week so I don't have my gear on me to try to reproduce 😞 I'll keep you posted!

noopkat, 2017-02-13:
@ts-ycc hi there, I could not reproduce this because it turns out I don't have an Arduino Pro Micro. I have an Arduino Micro, which is quite different.

You said your board is an Arduino __**Pro**__ Micro. If so, Avrgirl Arduino does not currently support this exact board yet. It has a slightly different bootloader, and is not the same as an Arduino Micro.

Does your board seem to be something like this? https://www.sparkfun.com/products/12640
Let me know, and if you're not sure feel free to attach a photo of your board to a reply 😄

ts-ycc, 2017-02-13:
Yes, indeed I have an Arduino Pro Micro. Oh well. :(  Thanks for the info, though.  Also, one thing I noticed is that the USB/serial driver (by Auduino) I used did not send an event when the board was reset and that might be the reason why the software hangs after it tried to reset the board.

noopkat, 2017-02-13:
ah I see. Could you try something for me? I wonder if you need the 'double reset' due to the slightly different bootloader. See this that I found: https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/32U4Note.pdf

if you were already resetting the board twice I apologise. I think there also might be a different flashing method as well, but I can ready the datasheet of the board and see if I can support this board with minimal effort.

ts-ycc, 2017-02-14:
I tried the double reset but it did not work, then I uninstall the current Arduino driver and reinstall another USB Serial driver and voila it works!!! The only thing I am not quite sure is the board does not blink and I used blink.cpp.hex to flash it. Here is the output:

No port auto sniff
1st list [ { comName: 'COM6',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_8037&MI_00\\7&14380AD2&0&0000',
    locationId: '0000.001d.0000.001.001.000.000.000.000',
    vendorId: '2341',
    productId: '8037',
    _standardPid: '8037' } ]
found micro on port COM6
resetting board...
reset complete.
connected
flashing, please wait...
flash complete.
null
2nd list [ { comName: 'COM7',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: undefined,
    pnpId: 'USB\\VID_2341&PID_0036\\6&796A6A0&0&1',
    locationId: 'Port_#0001.Hub_#0003',
    vendorId: '2341',
    productId: '0036',
    _standardPid: '0036' } ]

Looks good!! So the reason it did not work is because of the driver, but why the Arduino IDE works with the same driver is a bit puzzling. Also, you may want to add a timeout after the board is reset so for whatever reasons the driver does not respond the software will not hang. Thanks!

noopkat, 2017-02-15:
Glad you got it to work! Weird.
I do have a timeout in the code for after a reset, so I wonder if there's something I need to catch before the reset completes. Thanks