noopkat/avrgirl-arduino #91: Debug definition as function.

GermanBluefox, 2016-10-22:
Very nice package. 
I want to integrate it into my home automation platform ioBroker and found small problem.

It would be nice to have better integration possibilities. One of that is debug not to the console, but somewhere else.

`this.debug = options.debug ? (typeof options.debug === 'function') ? options.debug : console.log.bind(console)) : function() {};`

noopkat, 2016-10-30:
Hi @GermanBluefox! 

This is an awesome idea, I never considered the use case for this, but it makes a lot of sense.

One quick question before I code up a solution - from the code snippet you helpfully provided, am I interpreting it correctly to say that you would prefer an option to pass in your own debug function to do custom work with the debug strings every time `this.debug` is called?

Thanks! 😁

GermanBluefox, 2016-10-31:
Yes. You are right. I would like to pass my own debug function like 
```
function () {
  console.log.apply(console, arguments);
}
```

noopkat, 2016-11-06:
Hi again,
Gotcha, thanks for clarifying! I have published this feature request in the latest version of avrgirl-arduino. Let me know what you think, and if it helps you in the right way 😁

closed 2016-11-06 by noopkat

GermanBluefox, 2016-11-23:
Looks good. Thank you.