noopkat/avrgirl-arduino #96: Ability to specify custom board properties, or have the flashing process disregard 'incorrect' productIds

Mike-Dax, 2016-11-11:
We're rolling our own hardware which has specs similar to the leonardo, so when flashing with the Arduino IDE we'll select that option. However we have a different productId to what's expected by the library, as specified in boards.js. Obviously with this set 'incorrectly' we'll hit `Error: could not reconnect after resetting board.` since the productId doesn't match what's expected.

It'd be great if we could pass an object when creating the Avrgirl instance that specifies the board properties (to pass our modified productId), or alternatively have the search process just check that it can connect to the comName specified in the 'port' property of the options object.

noopkat, 2016-11-22:
Hi @Mike-Dax!

Thanks for raising this issue. I totally see where you're coming from, and why things are breaking. Thanks fir the thorough explanation, it was really helpful! I think allowing a custom board object would be the most flexible way going forward to resolve this.

This is a reasonable use case, so I published a new version which allows for a custom board object to be passed in, in place of the string board name.

I haven't had time to document it yet, but [there is an example in a demo file you can look at](https://github.com/noopkat/avrgirl-arduino/blob/master/tests/demos/micro.js#L2-L13) for getting started with custom boards. That Arduino micro example is a leonardo board, so your object contents would look somewhat similar.

Let me know if this works for you! 😁

noopkat, 2017-07-04:
closing this for now, comment if this wasn't resolved for you 😃

closed 2017-07-04 by noopkat