noopkat/avrgirl-arduino #107: Command not found

jmbaron, 2017-02-22:
I used the command `sudo npm install avrgirl-arduino` but when I try running the command `avrgirl-arduino list` I get an error saying `-bash: avrgirl-arduino: command not found` I ran `npm init` before running the first command and it still doesn't work.

noopkat, 2017-02-28:
Hi @jmbaron, thanks for raising this issue. It looks like you're attempting to use avrgirl-arduino globally, as a command line tool. If you run the same install command with the `-g` flag, this will install it as a global CLI app, instead of just for local code use. 

Try running: `npm install -g avrgirl-arduino` and if you get permission issues, you can reference [this documentation to fix it](https://docs.npmjs.com/getting-started/fixing-npm-permissions).

I hope that makes sense! Let me know how you go.

noopkat, 2017-03-19:
I'm closing this for now, you're welcome to reopen this if the issue is still persisting

closed 2017-03-19 by noopkat