noopkat/avrgirl-arduino #110: Pull request on noopkat/js-stk500v1

tocalvo, 2017-04-05:
Hi :wave:, i send a pull request to https://github.com/noopkat/js-stk500v1/pull/1 , because we have a problem with the console.log and verifyPage function. I think that maybe you don't see the pull requests on that repo. It's a small change to avoid problems when we use avrgirl-arduino inside a chrome extension. 
Thanks ! :smile:

sandeepmistry, 2017-04-10:
Hi @tocalvo,

Do you have an example app that can be used to reproduce the issue and verify your proposed fix?

noopkat, 2017-04-10:
Hi @tocalvo I have replied to the PR on noopkat/js-stk500v1#1 
In the meantime I'll pull this down to try to reproduce in that version of Linux.

tocalvo, 2017-04-17:
Hi @sandeepmistry , thanks for your help, i response in the js-stk500v1 fork, about the example app, the code its here: https://github.com/bq/bitbloq-chromeapp , to reproduce it, you can install the extension from here: https://chrome.google.com/webstore/detail/bitbloq/ddpdpmibfdhifigignfdiggfbcmfblfj and i just get the error on a specific educational distro called Guadalinex Edu , you can download it from here: http://www.juntadeandalucia.es/educacion/cga/portal/?page_id=7613 .
The easy way to check this, its creating and account on [Bitbloq](http://bitbloq.org) and activate the online compilation to force the use of the chrome extension.

These are some screenshots of the error:

[image]

[image]

Thanks!!

tocalvo, 2017-05-16:
Hi! Just a reminder for the pull request on the stk500 (the error detected on the PR was solved :D )
https://github.com/noopkat/js-stk500v1/pull/1
And sorry to write about this here. I dont want to bother, its just because the last time you can't see PR on your js-stk500v1 fork.  
Thanks!

closed 2017-05-21 by noopkat