noopkat/avrgirl-arduino #119: Error: could not open board on ...

christianalares, 2017-05-23:
I'm trying to flash a (working and tested) hex file to a Pololu Zumo 32U4.
The first time it get stuck on:
```found zumo on port /dev/cu.usbmodem1411
reset complete.
connected```

When I reload the page (that forces the back end to try again)  I get:
```found zumo on port /dev/cu.usbmodem1411
reset complete.
Error: could not open board on /dev/cu.usbmodem1411
    at /Users/krille/Dropbox/Dev/robotkodarn/node_modules/avrgirl-arduino/lib/connection.js:152:15
    at next (/Users/krille/Dropbox/Dev/robotkodarn/node_modules/awty/index.js:39:11)
    at SerialPort.<anonymous> (/Users/krille/Dropbox/Dev/robotkodarn/node_modules/avrgirl-arduino/lib/connection.js:143:7)
    at SerialPort._error (/Users/krille/Dropbox/Dev/robotkodarn/node_modules/serialport/lib/serialport.js:148:14)
    at SerialPort.<anonymous> (/Users/krille/Dropbox/Dev/robotkodarn/node_modules/serialport/lib/serialport.js:172:19)```

Note that I'm running the `manualReset` to `true`.

And by the way, the reason it says `zumo`is because I've made a custom board:
```var board = {
    name: 'zumo',
    baud: 57600,
    signature: new Buffer([0x1e, 0x95, 0x87]),
    productId: ['0x0036', '0x8036', '0x800c', '0x8036'],
    protocol: 'avr109'
}

var avrgirl = new Avrgirl({
    board: board,
    debug: true,
    manualReset: true
});```

I've also tried using the `leonardo` that should be very alike and may also work (it works using the native Arduino app when compiling and uploading) but with no success.

Any ideas?

noopkat, 2017-07-04:
Hi @christianalares sorry for the really slow response. This is curious 🤔 
I'll grab a zumo 32u4 and see if I can debug this. Looks like you're on OSX, is this correct? Thanks!

christianalares, 2017-07-05:
@noopkat Awesome!
That's right, I'm on OSX (or macOS as the cool kids say :). macOS Sierra to be precise.

pstenstrm, 2017-12-20:
Hi, I've managed to flash the Zumo32U4 robot by following the discussion in issue #84.

I set up Avrgirl with `manualReset` set to `false`, and when the first flash fails I immediately try to flash it again. What I noticed is that the `productId` changes.

The second time the board is in "bootloader mode". At first, after flashing the board with the Arduino IDE it has the `productId` `0x2300` and the bootloader has the `productId` `0x101`. So I set up a custom board with these values: 

``` 
zumo: {
    name: 'zumo',
    baud: 57600,
    signature: new Buffer([0x1e, 0x95, 0x87]),
    productId: ['0x0036', '0x8036', '0x800c', '0x8036', '0x2300', '0x101'],
    protocol: 'avr109'
  }
```

With this I am able to flash the board ONCE. After the flash the board will look like a `leonardo` board. Trying to flash it as `leonardo` will fail just as with the `zumo` settings, but again the flash immediately after will be in "bootloader mode".

By adding the bootloader `productId` to the `leonardo` board settings I'm able to flash it in this state as well:

```
leonardo: {
    name: 'leonardo',
    baud: 57600,
    signature: new Buffer([0x43, 0x41, 0x54, 0x45, 0x52, 0x49, 0x4e]),
    productId: ['0x0036', '0x8036', '0x800c', '0x101'],
    protocol: 'avr109'
  }
```

pstenstrm, 2017-12-20:
I think I figured out why the board changes to `leonardo`, I compile the sketch with avr-pizza-service and to make it easier for myself I set the board to `leonardo` instead of figuring out how to do it properly with the `zumo32u4` board. So the second part of my previous comment is probably only due to my short-cuts.

closed 2021-01-10 by noopkat