noopkat/avrgirl-arduino #124: Semantic Release

Apidcloud, 2017-05-28:
As mentioned in the stream, what do you think of using semantic-release in this repository?

I thought about the following way:
- [semantic-release] to handle automatic publishing on [github] and [npm][package]
- [commitizen] together with [cz-conventional-changelog] to help with semantic commits
- [husky] to handle pre-commits and commit-msg hooks together with [validate-commit-msg].

Hooks could be run optionally by default ([opt-in]) and therefore be easier for new contributors to actually contribute.

Let me know your thoughts :)

[semantic-release]: https://github.com/semantic-release/semantic-release
[github]: https://github.com/noopkat/avrgirl-arduino/
[package]: https://www.npmjs.com/package/avrgirl-arduino
[semantic-release]: https://www.npmjs.com/package/semantic-release
[commitizen]: https://www.npmjs.com/package/commitizen
[cz-conventional-changelog]: https://www.npmjs.com/package/cz-conventional-changelog
[validate-commit-msg]: https://www.npmjs.com/package/validate-commit-msg
[husky]: https://www.npmjs.com/package/husky
[opt-in]: https://www.npmjs.com/package/opt-in

noopkat, 2017-06-11:
@Apidcloud hey! Thank you for the suggestion. This would improve the collaborator/contributor experience, not to mention my own time that would also be saved. I'm going to put this on the roadmap to work on. ✨ 

If anyone else wants to pick up this as a task, they are most welcome to.

mcdonnelldean, 2017-06-11:
Just a note, mentioned on the stream. Happy to take these on. I'll start with Husky and maybe a couple of tests.

Apidcloud, 2017-06-11:
@mcdonnelldean let me know which of them you are taking on

mcdonnelldean, 2017-06-11:
@Apidcloud Husky is added. I haven't added the others but I'm happy to take a look or I can leave it to you too.

Apidcloud, 2017-06-11:
@mcdonnelldean I guess it's better if only one person works on this, as they are all interrelated. Just noticed the husky PR, so you can continue 👍 I'll be happy to help out either way!

mcdonnelldean, 2017-06-11:
@Apidcloud 👍

mcdonnelldean, 2017-06-11:
@Apidcloud So I had a look over both Semantic Release and Commitizen.

I'm gonna go ahead and leave those out of my PR and work. Basically I disagree with both and as such would be the wrong person to add these. That is of course not to say they are bad or wrong or anything like that; it is more I am of the belief things like that make the library harder to maintain for the maintainer as they (the maintainer) tend to have a "flow" that I feel these may interrupt.

With that in mind, it may make sense for you (or somebody else of course) to take a look at both with a view to PRing? For clarity, this is not my Lib, and I am not a major collaborator so please don't take this as a "This is bad, don't do it".

My recommendation is maybe do both on single PR's. While it will be a little ickier for you to do it that way, it will make @noopkat's life easier. Again for clarity, I'm not saying they are bad, I'm saying that I personally find both add maintainer complexity, and as such are not features I personally would like to add.

Apidcloud, 2017-06-12:
@mcdonnelldean I've come across them recently, and I think it's really helpful in the sense that besides automating the release process (on both npm and github), it also uses the (semantic) commit messages to produce the change log.

I didn't really explain the idea thoroughly, so I will try to do it below.

Personally, I like to follow git commit message rules (imperative verb and so on), and commitizen makes that really strict. It could also prove to be an extra burden, so the process can be triggered with a **commit** script (i.e., npm run commit) and **not by default** on committing. 
It would be something to help other people follow the same rules without really having to learn or remember them. 

Most of the times I don't really run the script, as I use `validate-commit-msg` (also with `opt-in`) on **commitmg** hook.

But I agree with you: @noopkat should be the one deciding if it's relevant for the repository at all. At the end of the day, those were all suggestions 💃

mcdonnelldean, 2017-06-12:
@Apidcloud Thanks for the additional info.

Yup, to stress, it's not at all that I think they are bad, more that they are things I personally wouldn't add. I still think you should perhaps make some PR's (showing is always better than telling). 

@noopkat Did seem to want to at least see these in action. I don't think that has changed. I just wanted to sound out that I didn't want to take those two parts on (I'd rather focus my own efforts on tests and coverage).

Having said that I would like to see the PR's too :D PR's are great ways to convince folk to change their ways :D

I noticed you said that it may make sense to do them all together including Husky. If that's the case please go ahead and do it that way. There is no rush to merge #128 and if it gets superseded by your PR it's all good. My PR is tiny so I won't be offended if something more robust is put in place.

Apidcloud, 2017-06-12:
@mcdonnelldean Aight 👍 Husky would just complement it, but it's not really necessary to make it work.

noopkat, 2021-01-10:
Gonna close this as it's pretty old and in the end I think that some of the frameworks here are now a bit overkill given how small this project still is and the number of contributors currently maintaining the project. Thank you for this work it is still appreciated 😃

closed 2021-01-10 by noopkat