noopkat/avrgirl-arduino #134: Error ocurred on CLI mode

lukywong, 2017-06-28:
I used blew command to flash a demo, but error comes
`avrgirl-arduino flash -f Blink.ino.eightanaloginputs.hex -a nano`

/Users/xxx/.nvm/versions/node/v6.10.2/lib/node_modules/avrgirl-arduino/bin/cli.js:40
      } else if (!boards.byName[argz.a]) {
                               ^

TypeError: Cannot read property 'nano' of undefined
    at handleInput (/Users/xxx/.nvm/versions/node/v6.10.2/lib/node_modules/avrgirl-arduino/bin/cli.js:40:32)
    at Object.<anonymous> (/Users/xxx/.nvm/versions/node/v6.10.2/lib/node_modules/avrgirl-arduino/bin/cli.js:109:1)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:393:7)
    at startup (bootstrap_node.js:150:9)

noopkat, 2017-06-29:
@lukywong hi there! Thanks for reporting this bug. It was a regression from a change made recently. I have patched and released a new version to npm, could you please try installing the latest avrgirl-arduino and running the command again? 

Apologies for the inconvenience and thanks again for reporting this!
[image]

lukywong, 2017-06-29:
it works very well now, thank you

closed 2017-06-29 by lukywong