noopkat/avrgirl-arduino #139: The latest version(2.2.7) can't flash successfully on mega2560

focalhot, 2017-12-06:
The latest version(2.2.7) can't flash on mega2560，but 2.2.4 can flash successfully on mega2560,Do anyone else have such a problem？

```
var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'mega',
  debug: true,
});

avrgirl.flash('./hex/focalhot.hex', function (error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});
```

obra, 2017-12-06:
what sort of error are you getting?

noopkat, 2018-01-14:
@focalhot hi, thanks for reporting this issue. I was able to reproduce this. For me, the process hangs after resetting the board. Does this also happen for you?

This is what I see in my console:

```
$ > node tests/demos/mega.js
found mega on port /dev/tty.usbmodem146211
connected
reset complete.
|
```

I have tracked it down to the Serialport 5 upgrade commit, and will investigate a dependency of this library to track down exactly what is causing this before potentially opening an issue on the serialport library. Just gotta cross my t's and dot my i's first to ensure it's not an adjustment I need to make on this end.

In the meantime, I think it might be best to roll back the Serialport version if I can't easily release a patch. Please use v2.2.4 for now and I'm sorry for the inconvenience 🙇‍♀️

focalhot, 2018-01-19:
@noopkat Serialport v2.2.4 can work well for me,thanks.
Here is my code:
`
  "dependencies": {
    "avrgirl-arduino": "^2.2.4",
    "serialport": "^4.0.1",
    "debug": "^2.6.0",
    "iconv-lite": "^0.4.15",
    "request": "^2.79.0",
    "underscore": "^1.8.3"
  }
`

noopkat, 2018-01-28:
Quick update - I have reverted the serialport dependency upgrade and is now back to installing and using v4.x. This has been published in avrgirl-arduino v2.2.9.

@obra feel free to stay on v2.2.8 for now, as I'll be looking into this issue and hopefully will have a thorough fix for this and can upgrade serialport again very soon.

[image]

closed 2018-01-28 by noopkat

ta2edchimp, 2018-06-03:
The `serialport` module does not fire a `close` event (anymore) without reading from or writing to the connection.  
In a small test project of mine, where I lent the resetting portion from avrgirl-arduino, I was able to work around this issue by also registering for `data` events prior to setting up a listener for the `close` event.  
Maybe this can be of use some time later.