noopkat/avrgirl-arduino #140: ESP32/ESP8266 support?

vshymanskyy, 2018-01-02:
These boards are widely used an popular, are there any plans to add support fort them?
Yeah, I understand it's not AVR family, but anyway..

x4m3, 2018-02-25:
yeah that would be awesome, i just bought a [lolin esp32](https://wiki.wemos.cc/products:lolin32:lolin32), maybe i'll add the board to avrgirl! 

@noopkat is there a guide on how to add a new board ?

noopkat, 2018-03-01:
Hi @vshymanskyy thanks for opening this issue about esp32/8266 support.

AVRGirl does indeed only support the AVR family of devices at present. It is a pure JavaScript implementation of the protocols used by the different boards.

The ESP 8266/32 boards can run Arduino code in running mode, but has a different bootloader to AVR chips when programming/flashing the sketch to the device. There's a variety of flashing tools out there, such as [esptool](https://github.com/espressif/esptool), [nodemcu flasher](https://github.com/nodemcu/nodemcu-flasher), and [nodemu py-flasher](https://github.com/marcelstoer/nodemcu-pyflasher). 

I'd prefer we avoid importing these existing tools and having a NodeJS implementation of them. So it's possible, but it's some work to put on the roadmap 😄 

We'd want to:

1. decide on which bootloader to support (probably nodemcu for now)
2. implement a NodeJS version of the flasher tool 
3. integrate this new 'protocol' support in AVRGirl
4. develop and implement a new schema for the esp boards (you can look at boards.js in the root of this repo directory for the current AVR flavoured schemas)

@x4m3 there isn't currently a guide on how to add a new board, but I will definitely open an issue to address this. Great idea on having a guide. Currently, it's not necessary to add a new board to this repo to use it. AVRGirl supports passing in an object (instead of a string, like "uno") for the board property in the options when creating the instance of avrgirl. This should also be documented but it is not currently 😄 

Let's see what we can do. Right now I'm in the middle of a large feature to implement progress events (which has been needed for a long time) but happy to circle back on this to discuss further once this feature ships 👍

ruudboon, 2020-07-12:
+1 for me

noopkat, 2020-12-20:
Hey folks,

After a lot of thought, I can't commit to providing support for ESP-x devices in this repo. It would add more dependency risk burden because of the tetchy nature of native hardware bindings. This library is pretty tightly scoped to only support Atmel AVR type microcontroller breakouts, and ESP breakouts are a totally different architecture / toolchain. So it doesn't make contextual sense to support these breakouts in avrgirl-arduino. This library has always only ever had a single maintainer despite some helpful part time contributors popping in and out sporadically, so it's also going to be too much scope addition for me to handle given the limited time I have to work on open source these days.

I hope this makes sense and I'm sorry for not addressing this sooner.

closed 2020-12-20 by noopkat