noopkat/avrgirl-arduino #141: create a Github issue template

noopkat, 2018-01-14:


jonasbn, 2018-01-14:
I can take time to make a first shot at this

chriszinkula, 2018-01-15:
I went ahead and tried my hand at it.  https://github.com/noopkat/avrgirl-arduino/pull/143

jonasbn, 2018-01-15:
My proposal is available as PR #144 

I have also added a PR template. I like @chriszinkula's 

- Expected behaviour
- Actual behaviour

For the issue template, I have these in my own issue templates, but have not added these to my proposal at this time, since my proposal for the issue template focus on assisting in getting the user to provide factual data about the environment. But it could be integrated easily.

For the PR template large parts of the issue template could be added, but I will let you decide on what makes sense and provides value in regard to obtaining information on hardware as metnioned in PR #144

noopkat, 2018-01-21:
closing as a result of #143 and also #144 thanks so much @chriszinkula and @jonasbn 💯

closed 2018-01-21 by noopkat