noopkat/avrgirl-arduino #145: Upgrade serialport dependency

noopkat, 2018-01-28:
## Description

Latest version of Serialport does not work with this library. Not sure if serialport end or stk500v2 end or avrgirl-arduino end.

Going to investigate and fix this as soon as possible! This issue is tracking this work.

[image]

phptuts, 2018-04-13:
So I figured out what was going on.  It's that SerialPort.list returns a promise.  I created a fix for it on fork, but I could not fix the tests.

https://github.com/phptuts/avrgirl-arduino/blob/master/lib/connection.js#L186

phptuts, 2018-04-13:
Hope this helps, sorry could not fix the tests.

noopkat, 2019-12-03:
closing as we have upgraded to serialport 8 at this stage

closed 2019-12-03 by noopkat