noopkat/avrgirl-arduino #146: Is it possible to get flashing progress (percentage)?

altjz, 2018-01-29:
First. Many thanks to this great library. It save a lot of time.
And forgive me the issue template doesn't fit the feature request.

## Feature request
Provide flashing progress.

## Expected behavior
may be add progress callback here
```javascript
var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'uno'
});

avrgirl.flash('Blink.cpp.hex', function (error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
}, function(progress) { // progress here
    console.log(progress);
});
```

## Description
I used `avrdude` to flash the board before. 
Avrdude doesn't have the progress, But I can detect the output string with symbol '#' to calculate progress.
The information of debug mode is not enough. Any way to get progress on avrgirl-arduino ?

### Current debug mode
```
found nano on port /dev/cu.usbserial-A50285BI
connected
reset complete.
flashing, please wait...
flash complete.
```
### Avrdude
```
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/alex/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem14B431
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/var/folders/dl/l61g8nsd3cbdt05jk2xjrdt40000gp/T/arduino_build_896654/StandardFirmata.ino.hex"
avrdude: writing flash (12330 bytes):

Writing | ################################################## | 100% 2.24s

avrdude: 12330 bytes of flash written
avrdude: verifying flash memory against /var/folders/dl/l61g8nsd3cbdt05jk2xjrdt40000gp/T/arduino_build_896654/StandardFirmata.ino.hex:
avrdude: load data flash data from input file /var/folders/dl/l61g8nsd3cbdt05jk2xjrdt40000gp/T/arduino_build_896654/StandardFirmata.ino.hex:
avrdude: input file /var/folders/dl/l61g8nsd3cbdt05jk2xjrdt40000gp/T/arduino_build_896654/StandardFirmata.ino.hex contains 12330 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.67s

avrdude: verifying ...
avrdude: 12330 bytes of flash verified

avrdude done.  Thank you.
```

noopkat, 2018-03-01:
Hi @altjz thanks for opening this issue.

Progress reporting is a feature I have wanted to implement for a long time now. After this issue was opened by you I started working on this. I'll set up a milestone and reference this issue soon so you can track progress on this feature. It is a big job to implement for a variety of reasons, but we're getting there slowly 😄 

Would you be interested in helping test this feature once it's ready?

Thanks again

altjz, 2018-03-06:
Hi @noopkat, thanks for response.
I can't wait to test this feature! Let me know once it's ready! 😃

s00500, 2019-07-10:
Hi @noopkat, what is the status of this ? Is there any work that can be contributed?

noopkat, 2019-08-04:
@s00500 I'm about halfway through implementing this but I want to prioritize the work on my end to get it done asap. Sorry it has taken so long it's been a lot of work.