noopkat/avrgirl-arduino #150: Can't install, permission denied

alexgleason, 2018-02-13:
Hello! I'm trying to `sudo npm install -g avrgirl-arduino` and running into issues.

Node: v9.5.0
npm: 5.6.0
OS: elementary OS 0.4.1 Loki 64-bit (based on Ubuntu 16.04.3 LTS), Linux 4.4.0-112-generic

Here's my full journey:

```
alex@Starman:~/Downloads$ sudo npm install -g avrgirl-arduino
npm ERR! code 1
npm ERR! Command failed: /usr/bin/git clone --depth=1 -q -b api-updates https://github.com/noopkat/browser-serialport.git /home/alex/.npm/_cacache/tmp/git-clone-3981b634
npm ERR! /home/alex/.npm/_cacache/tmp/git-clone-3981b634/.git: Permission denied
npm ERR! 
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/alex/.npm/_logs/2018-02-13T23_45_26_660Z-debug.log

alex@Starman:~/Downloads$ sudo rm -rf ~/.npm/_cacache/

alex@Starman:~/Downloads$ npm install -g avrgirl-arduino
npm WARN checkPermissions Missing write access to /usr/lib/node_modules
npm ERR! path /usr/lib/node_modules
npm ERR! code EACCES
npm ERR! errno -13
npm ERR! syscall access
npm ERR! Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!  { Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!   stack: 'Error: EACCES: permission denied, access \'/usr/lib/node_modules\'',
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/lib/node_modules' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/alex/.npm/_logs/2018-02-13T23_46_25_728Z-debug.log

alex@Starman:~/Downloads$ sudo npm install -g avrgirl-arduino
/usr/bin/avrgirl-arduino -> /usr/lib/node_modules/avrgirl-arduino/bin/cli.js
> serialport@4.0.7 install /usr/lib/node_modules/avrgirl-arduino/node_modules/serialport
> node-pre-gyp install --fallback-to-build
node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@9.5.0 | linux | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v59-linux-x64.tar.gz
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v59-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@9.5.0 (node-v59 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http EACCES: permission denied, mkdir '/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build' 
node-pre-gyp verb command build [ 'rebuild' ]
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build'
gyp ERR! System Linux 4.4.0-112-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release"
gyp ERR! cwd /usr/lib/node_modules/avrgirl-arduino/node_modules/serialport
gyp ERR! node -v v9.5.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:160:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:943:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
node-pre-gyp ERR! System Linux 4.4.0-112-generic
node-pre-gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /usr/lib/node_modules/avrgirl-arduino/node_modules/serialport
node-pre-gyp ERR! node -v v9.5.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/usr/lib/node_modules/avrgirl-arduino/node_modules/serialport/build/Release' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/alex/.npm/_logs/2018-02-13T23_46_46_773Z-debug.log

alex@Starman:~/Downloads$ sudo su

root@Starman:/home/alex/Downloads# npm install -g avrgirl-arduino
npm ERR! code 128
npm ERR! Command failed: /usr/bin/git clone --depth=1 -q -b api-updates https://github.com/noopkat/browser-serialport.git /root/.npm/_cacache/tmp/git-clone-0f23dc62
npm ERR! fatal: could not create leading directories of '/root/.npm/_cacache/tmp/git-clone-0f23dc62': Permission denied
npm ERR! 
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-02-13T23_47_36_653Z-debug.log

root@Starman:/home/alex/Downloads# rm -rf ~/.npm

root@Starman:/home/alex/Downloads# npm install -g avrgirl-arduino
npm ERR! code 128
npm ERR! Command failed: /usr/bin/git clone --depth=1 -q -b api-updates https://github.com/noopkat/browser-serialport.git /root/.npm/_cacache/tmp/git-clone-62fa1559
npm ERR! fatal: could not create leading directories of '/root/.npm/_cacache/tmp/git-clone-62fa1559': Permission denied
npm ERR! 
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-02-13T23_48_13_340Z-debug.log

root@Starman:/home/alex/Downloads# 
```

Any ideas? Thanks!

noopkat, 2018-02-14:
Hi @alexgleason, sorry that you're running into dramas getting avrgirl-arduino installed.

It looks like it's getting stuck building the native dependencies within node-serialport (which avrgirl-arduino relies on for serial communication). Could you please try the following to see if you get the same errors?

```
sudo npm install -g serialport
```

Thanks! 🙇

noopkat, 2019-12-03:
this issue is a little dated now so I'm going to close it, but please feel free to re-open if you're still having issues. Thanks!

closed 2019-12-03 by noopkat