noopkat/avrgirl-arduino #156: Tried to download(404): node-serialport

AlparLeonardoMeyer, 2018-04-09:
# Issue Template

Error on install this module.

## Description

Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v59-win32-x64.tar.gz

### Expected behaviour

_What behaviour did you expect did occur?_

### Actual behaviour

_What unexpected behaviour occurred?_

## Operating system and version

_Please provide version on the operating system etc. used._

### Windows (pre version 7)

`C:\> ver`

### Windows (Version 7 and newer)

`C:\> systeminfo | findstr /B /C:"OS Name" /C:"OS Version"`

### Linux and Unix based systems

`$ uname -a`

_For information on the specific distribution. Please consult your distribution documentation._

## Avrgirl Arduino version

_Please provide version on the Avrgirl Arduino version used_

_If working in a fork of the GitHub project, obtain this information post installation using:_

`$ npm list | grep avrgirl-arduino`

_If you are working on code, which has diverged from the lastest release, please mention this_.

## NodeJS version

_Please provide version on the Node version used._

`$ node --version`

## Arduino Board being used

_Please provide version/model on the Arduino board being used._

_A full List of boards and processor names are [available on Wikipedia](https://en.wikipedia.org/wiki/List_of_Arduino_boards_and_compatible_systems)._

## Log output, if available

_Please add any log output, which can be used to describe the issue or circumstances surrounding the issues._

## Step by step guide to reproducing the issue

_If possible a code snipppet to demonstrate the issue or a throught step by step description of how to get reproduce the error._

noopkat, 2018-04-09:
Hi there, thanks for opening this issue. This is not a bug related to avrgirl-arduino, rather a bug on [node-serialport](https://github.com/EmergingTechnologyAdvisors/node-serialport). Could you kindly open an issue on their repository? Thanks!

ajfisher, 2018-07-05:
@noopkat I've just run into this on a couple of my repos as well. Seems like SerialPort 4 hasn't been ported to Node 10 and newer MacOS as well.

I've done a quick POC on avrgirl-arduino and if you switch out `serialport` for v6 it appears to work okay. Do you want me to raise that change as a PR?

noopkat, 2019-12-03:
this issue is now both stale and additionally we're now in serialport 8 so we should be good to close this 😄

closed 2019-12-03 by noopkat