noopkat/avrgirl-arduino #173: arduino leonardo flash hex not work

wuyinyao, 2019-06-01:
i try to upload hex to arduino leonardo board, always get could not open xxx port error, as i know the com port no will change when upload hex, so i think that is root cause.

right now, i just modify the connection.js code make the com port correct when upload hex the com port change,  not show "cound not open xxx port error" , and see the leonardo board upload hex is working-the indicator keep flashing, but occur error when  log print programming xxx bytes.

you guys meet this problem ever, pls offer some clue to fix this problem, thanks!

wuyinyao, 2019-06-01:
run in win7

noopkat, 2019-06-02:
hi @wuyinyao can I ask if you were manually specifying the first port (before reset) in your code like below?

```javascript
var avrgirl = new Avrgirl({
  board: 'uno',
  port: '/dev/cu.usbmodem1412'
});
```

if you are, then this might be causing the bug. If not, the connection file code should still be trying to autofind the board, and won't necessarily try to reconnect to the same com port. This is because I'm aware that port switching is a thing that happens on windows with leonardo boards, so I built in an auto find function in that sequence of code: https://github.com/noopkat/avrgirl-arduino/blob/master/lib/connection.js#L110

Does this make sense?

noopkat, 2019-12-08:
closing this as it's a stale issue but please feel free to reopen if this is still plaguing you 🙇‍♀

closed 2019-12-08 by noopkat