noopkat/avrgirl-arduino #174: add url to product page for each board

noopkat, 2019-06-14:
# Issue Template

Please review the [contribution](https://github.com/noopkat/avrgirl-arduino/blob/master/CONTRIBUTING.md) and [code of conduct](https://github.com/noopkat/avrgirl-arduino/blob/master/CODE_OF_CONDUCT.md) guidelines.

## Description

We should introduce a new property to the boards.js list which is a string url to the product page of each board, suggested by @nnja ✨ 

## Why?

This would be really handy for both contributors to understand what all of the boards are, and also for helpful error output if a board cannot be autosniffed (we can ask the user to verify that specific board with a link to the product page)

Lisiadito, 2019-08-11:
With product page you mean this for example [Store Arduino UNO](https://store.arduino.cc/arduino-uno-rev3) ?
I would be willing to do that.

noopkat, 2019-08-11:
@Lisiadito totally! Let me know if you run into tricky issues sourcing the right URLs. Some of the boards we support in avrgirl are no longer in circulation / available to buy. An info page would suffice if a buy link doesn't work.

Thank you! 😄

Lisiadito, 2019-08-12:
@noopkat I've found for most boards a page (buy or info). But I can't find a info page about the old ardunio nano.

noopkat, 2019-08-12:
@Lisiadito this is exactly the board I was thinking would give you the trouble 😂 
I'll dig around for a suitable replacement. Thanks for the update!

noopkat, 2019-08-12:
@Lisiadito ok let me know if this is a silly idea to you, but I found the [original Nano hardware info page](https://web.archive.org/web/20150813095112/https://www.arduino.cc/en/Main/ArduinoBoardNano) from the Internet Archive and think this would be the most accurate we can find and still from the official website?

Lisiadito, 2019-08-12:
@noopkat hahaha I did exactly the same yesterday but wasn't sure if that's the best solution. But if that's the only link we both find I guess it's the best shot. I will add it and make the PR

Lisiadito, 2019-09-29:
I think this can be closed since the PR is merged.

closed 2019-12-03 by noopkat