noopkat/avrgirl-arduino #175: Error on install on node 12.4.0

gorhgorh, 2019-07-05:
# Error on install on node 12.4.0

serialport version (v6.2.2) used in Avrgirl Arduino (v2.2.9) does not provide pre built binaries for node 12, and compiling does not work too (tried it in a workshop on multiple machine)

in the release note you mention reverting serialport because of some board issues (MEGA), so you may not want to have it updated, but the latest version of serial port DOES compile on 12.4.0.

Sidenote, fs-event, that some package in the the project uses too, have the same issue (not compiling with used version and node 12), so maybe just a disclaimer in the readme about osx and node 12.x would be enough ?

Everything works perfectly in node 10 on the same machine.

This issue affect firmata-party too (I discovered the issue with FP)

Please let me know if I can help in any way.

Thanks for these super tools that you are creating!

### Expected behaviour

it should install avrgirl-arduino globally when using node 12.4.0
### Actual behaviour

it does not succeed in compiling serial port preventing the installation to finish

## Operating system and version
osx 10.14.5

## Avrgirl Arduino version
Avrgirl Arduino  : 2.2.9

## NodeJS version
12.4.0

## Arduino Board being used

NA

## Log output, if available

i can make a pastebin if needed, but this related to serial port

## Step by step guide to reproducing the issue

- install node 12.4.x
- ```npm install -g avrgirl-arduino@latest```

noopkat, 2019-08-04:
Hi thanks for reporting this issue. I think this is to do with avrgirl needing an upgrade to the serialport dependency. I'll need to do this and test on all operating systems before publishing the compatibility updates. Thanks for your patience I should have this done really soon 🙇‍♀

gorhgorh, 2019-08-15:
hey hey, no worries here I'm using nvm and switch for using avrgirl/firmata party, and yes this has to do with old serial port version used, as mentioned I did not dared make a PR for that as I don't have the boards to test possible regressions.

Thanks again for all your cool work!

closed 2019-08-15 by gorhgorh