noopkat/avrgirl-arduino #180: Error: could not reconnect after resetting board.

paulhayes, 2019-10-02:
## Description

Uploading to any of my sparkfun pro micro clones doesn't work. These boards use official atmel chips, and work in Arduino and Platformio without problems. Here's a simple test I wrote that demonstrates the issue:

```
var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'sf-pro-micro'
});

avrgirl.flash('node_modules/avrgirl-arduino/junk/hex/sf-pro-micro/Blink-5v.cpp.hex', function (error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});
```

### Expected behaviour

Successful upload to sparkfun pro micro

### Actual behaviour

The program dumped the following error
```
Error: could not reconnect after resetting board.
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:133:23
    at next (C:\Users\Paul\Workspace\firmata-party\node_modules\awty\index.js:39:11)
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:118:7
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:72:12
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:211:12
    at promise.then.ports (C:\Users\Paul\Workspace\firmata-party\node_modules\serialport\lib\serialport.js:628:16)
```

## Operating system and version

OS Name:                   Microsoft Windows 10 Pro
OS Version:                10.0.17763 N/A Build 17763

_For information on the specific distribution. Please consult your distribution documentation._

## Avrgirl Arduino version

v3.0.2

## NodeJS version

v10.16.3

## Arduino Board being used

sparkfun pro micro 5v 16Mhz  clone

( Although this board is a clone, it works with platformio and arduino IDE's without issue, I've been using them for about 3 years without uploader issues )

## Log output, if available

_Please add any log output, which can be used to describe the issue or circumstances surrounding the issues._

## Step by step guide to reproducing the issue

_If possible a code snipppet to demonstrate the issue or a throught step by step description of how to get reproduce the error._

noopkat, 2019-10-05:
@paulhayes thanks for reporting this issue. There is a chance that this board has a different product id when it's in bootloader mode and I don't have it in my list to poll for once the board's USB serial emulation comes back online after resetting. 

Would you mind performing the following steps below to confirm this is the case or not:

1. Plug in the Pro Micro to your computer
2. Run the following code:

```js
var Avrgirl = require('avrgirl-arduino');

Avrgirl.list(function(error, list) {
  console.log('pre-reset list:', list);
});

var avrgirl = new Avrgirl({
  board: 'sf-pro-micro'
});

avrgirl.flash('node_modules/avrgirl-arduino/junk/hex/sf-pro-micro/Blink-5v.cpp.hex', function (error) {
  avrgirl.list(function(error, list) {
    console.log('post-reset list:', list);
  });
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});

```

If you could share the output of your terminal by copy + pasting it in this issue that would be helpful to debug. Feel free to filter out the other USB device info if you want to for privacy reasons.

I realise the above code is a little janky but it should get the output I need to debug. If the Pro Micro had a physical reset button this would be a little easier.

paulhayes, 2019-10-07:
Yup, I think you're right. Unusual product id seems to be the cause. Will you add support for this variant to your library?

Thanks so much for taking the time to address this issue.

```
pre-reset list: [ { comName: 'COM29',
    manufacturer: 'Microsoft',
    serialNumber: '7&22d08be5&0&1',
    pnpId: 'USB\\VID_1B4F&PID_9206&MI_00\\8&21D8619A&0&0000',
    locationId: '0000.0014.0000.003.003.001.000.000.000',
    vendorId: '1B4F',
    productId: '9206',
    _standardPid: '9206' } ]
Error: could not reconnect after resetting board.
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:133:23
    at next (C:\Users\Paul\Workspace\firmata-party\node_modules\awty\index.js:39:11)
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:118:7
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:72:12
    at C:\Users\Paul\Workspace\firmata-party\node_modules\avrgirl-arduino\lib\connection.js:211:12
    at promise.then.ports (C:\Users\Paul\Workspace\firmata-party\node_modules\serialport\lib\serialport.js:628:16)
post-reset list: [ { comName: 'COM31',
    manufacturer: 'Arduino LLC (www.arduino.cc)',
    serialNumber: '7&22d08be5&0&1',
    pnpId: 'USB\\VID_2341&PID_0036\\7&22D08BE5&0&1',
    locationId: 'Port_#0001.Hub_#0006',
    vendorId: '2341',
    productId: '0036',
    _standardPid: '0036' } ]
```

closed 2019-12-08 by noopkat

noopkat, 2019-12-08:
hi @paulhayes sorry for taking so long with this but I have added the product id to the latest release of this library. For future FYI and I should have told you this earlier (🙈) but you can supply a custom board object as outlined in the README in order to get around instead of waiting me to add it officially to the matching board configuration.

Can you kindly let me know if the latest release doesn't fix reset port hopping bug for you? Then I'll investigate further 🙇‍♀ 

[image]

commit 7ad8c8a8 (2019-12-08): fixes #180 add sp pro micro clone pid