noopkat/avrgirl-arduino #190: Web serial demo feedback

MichaelE1, 2020-01-25:
Hi Suz! I had a chance to mess around with the web serial demo today. It's working really well. The only issue I ran into was that a flag needs to be enabled in Chrome stable for web serial support. I've opened a [PR](https://github.com/noopkat/avrgirl-arduino/pull/189) in hope that it will save someone the time I spent debugging 😅 

I also had a go at quickly converting the demo to [React](https://github.com/MichaelE1/web-serial-react). The code is a bit messy, but the great news is that it also works. This is using the UMD AvrGirl Arduino library distribution as opposed to the global window version too!

No pic, no proof:

[image]

( :

noopkat, 2020-01-27:
Hi @MichaelE1 
Thank you SO MUCH for testing this. I lost it at the end with the waving sloth!!!!!! This is fantastic.

I would be happy to receive the React version of the demo as a pull request if you're comfortable with the code as is? Feel free to tweak it beforehand as you said it was messy and I'd want you to be happy with it. I'm thinking that we could have a vanilla javascript demo and then a react demo to help folks with getting up and running if they want to use an SPA framework approach. 

What do you think?

Thanks again I REALLY appreciate people testing this stuff out on their own machines. I actually had completely forgotten that I had enabled a Chrome flag for this to work so we should definitely add this to the readme in the demo directory 🙇

noopkat, 2020-01-27:
Edit: just saw you opened a PR already 🤦‍♂  I'll take a look at it

closed 2020-02-02 by noopkat