noopkat/avrgirl-arduino #193: Flashing some Avr109 devices does not reboot back to the user sketch

algernon, 2020-03-07:
## Description

We're using this wonderful library in [Chrysalis](https://github.com/keyboardio/Chrysalis) to flash firmware onto some of the supported keyboards. We recently upgraded our avrgirl-arduino dependency from v3.0.0 to v4.2.1, and noticed that flashing no longer works: the process itself finishes, verification works too, fuse checking passes as well, but the keyboard doesn't reset back to the newly flashed firmware, and stays in bootloader mode.

### Expected behaviour

The expected behaviour, which we had in v3.0.0, is that once flashing is done, the keyboard resets to the newly flashed firmware.

### Actual behaviour

The keyboard doesn't appear to reset, and gets stuck in bootloader mode.

## Operating system and version

The problem does not appear to be OS specific. It has been reproduced on Debian Stable, Ubuntu LTS, Windows 10 and macOS Catelina.

## Step by step guide to reproducing the issue

I've bisected the problem (see below), and used the following script for testing:

```js
var Avrgirl = require('./');
var board = {
  name: "Keyboardio Model 01",
  baud: 9600,
  productId: ["0x2300", "0x2301"],
  protocol: "avr109",
  signature: new Buffer.from([0x43, 0x41, 0x54, 0x45, 0x52, 0x49, 0x4e])
};

var avrgirl = new Avrgirl({
  board: board,
  debug: true
});

avrgirl.flash("default.hex", function(error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});
```

The `default.hex` referenced by the script is [this file](https://github.com/keyboardio/Chrysalis/raw/master/static/Keyboardio/Model01/default.hex).

Using a Keyboardio Model01, I started bisecting from v3.0.0, which worked, ran `npm i` and `node ./model01-test.js` each iteration, and checked whether the keyboard booted into the new firmware. This allowed me to narrow the problem down to aec77ea02033dfa0d848896f336da809db8748e4. And indeed, backing that change out, the keyboard now boots into the newly flashed firmware.

## Questions?

I'm not sure how to best tackle this problem, where and what fix is needed. I'm happy to provide any additional info if need be. In the meantime, would it be acceptable to make the closing optional, by a board property for example?

algernon, 2020-03-07:
FWIW, I quickly hacked up a patch for the board property mentioned above, it's [here](https://github.com/algernon/avrgirl-arduino/commit/7b21687fd4d034ce905ed95171b1703f7329e941). I'm happy to turn that into a PR, or rework it.

Unless, of course, we can figure out what's the root of the problem, rendering the above patch unnecessary. :)

noopkat, 2020-03-22:
Hi @algernon I am so sorry for taking so long to respond! Thanks for investigating this. The backstory is that this line of code (the one closing the connection) had been commented out for a really long time. I looked up the original issue which was in node-serialport, and it looked like it was now fixed. I tested this on Mojave and it seemed fine. I definitely should have tested this more robustly 🙇 

I think what I'll do is comment this line back out again and link to this issue this time, and let the node-serialport folks know in case they have bandwidth to investigate. I'll publish this as a patch and I'll follow up on this issue again to close it out once that's all done.

I'm really glad you found a workaround so quickly, and thanks for sharing it with me as it meant an instant understanding of the root cause (at least the surface level root cause!)

I appreciate your patience once again - I'm a real big fan of Keyboardio and all of you lovely folks over there.

noopkat, 2020-03-22:
patch released in v4.2.2 🙇

commit d2287d87 (2020-03-22): remove avr109 port close after flash complete again

temporary fixes #193
closed 2020-03-22 by noopkat