noopkat/avrgirl-arduino #195: Program not exiting flash function after successfully uploading hex file

msanatan, 2020-04-15:
## Description

I use macOS Catalina and was looking for an easy way to flash games on my [Arduboy](https://arduboy.com/) and came across your (awesome!) library. I'd like to use it to develop a new app to manage my game library, which is a bunch of hex files. 

As a start, I installed the NPM package in a folder and added this `index.js` which I adapted from your README:

```javascript
var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'leonardo',
  debug: true
});

avrgirl.flash('pong.hex', function (error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});
```

The hex file is attached as a zip, just because lol. While the script successfully flashes the hex file, the node program does not quit. This is the output I see in my console:

```console
marcussanatan@Marcuss-MacBook-Pro arduboy-upload % node index.js                        
resetting board...
found port on /dev/tty.usbmodem14201
reset complete.
connected
flashing, please wait...
flash complete.
done.
^C
marcussanatan@Marcuss-MacBook-Pro arduboy-upload % 
```

Note the `^C` as I have to manually interrupt the program to get back to my shell. The "done" messaged was logged to the screen, further confirmation that the flashing did not encounter an error. However, it seems as though the script is "stuck" in the callback.

I was curious if this occurred in earlier versions. I deleted `node_modules` and `package-lock.json` and installed version **4.2.1**. When I ran my script again, it just works:

```console
marcussanatan@Marcuss-MacBook-Pro arduboy-upload % node index.js                        
resetting board...
found port on /dev/tty.usbmodem14201
reset complete.
connected
flashing, please wait...
flash complete.
done.
marcussanatan@Marcuss-MacBook-Pro arduboy-upload % 
```

I'll try to deep dive when I have more time this week, but I took a look at the differences between 4.2.1 and 4.2.2. Most of it involves web serial updates except for d2287d87bbc6ba94040e4330a52cc867f684bf53 which is a patch for issue #193. Time permitting I'll confirm in the next couple of days but it feels like the culprit.

### Expected behaviour

I expected the Node.js program to quit after the hex file was flashed

### Actual behaviour

The Node.js program is blocked when I call the `flash()` function.

## Operating system and version

### Linux and Unix based systems

Darwin Marcuss-MacBook-Pro.local 19.3.0 Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64 x86_64

Haven't confirmed on other OSes.

## Avrgirl Arduino version

_Please provide version on the Avrgirl Arduino version used_

_If working in a fork of the GitHub project, obtain this information post installation using:_

avrgirl-arduino@4.2.2

## NodeJS version

v13.3.0

## Arduino Board being used

Arduboy (leonardo)

[pong.hex.zip](https://github.com/noopkat/avrgirl-arduino/files/4478611/pong.hex.zip)

msanatan, 2020-04-18:
Hey, just following up on this. I did a test where I uncommented this line in `lib/avr109.js`:
https://github.com/noopkat/avrgirl-arduino/blob/master/lib/avr109.js#L53

Results when running script

```console
marcussanatan@Marcuss-MacBook-Pro arduboy-upload % node index.js 
found leonardo on port /dev/tty.usbmodem14201
resetting board...
found port on /dev/tty.usbmodem14201
reset complete.
connected
flashing, please wait...
flash complete.
done.
```

Works just like in 4.2.1. I know it may be the fix for the issue in the comments, but it seems to be affecting other behavior.

noopkat, 2020-05-03:
Hi @msanatan,
Thanks for flagging this, as I was interested in whether or not this is something that affects how people use the library for their projects.

Just for history, this library has always had this bug until 4.2.1 as you were able to track down, as you mentioned due to #193. Given that this bug is still going to be a thing going forward for certain operating systems, I need to work in a more sophisticated fix. In the meantime, I would recommend just using v 4.2.1 for now especially if everything else in that version is working as expected.

Stay tuned for a proper fix soon, likely in a version 4.2.3?

BTW it's really cool to see that you're using this for the Arduboy! I love that board a lot and am happy to hear that people are still using it. It's always kinda needed an easy ROM loader, so I'm glad you worked on a solution that works for you 😄 

Thanks again for your kind words and generosity towards this project! 💯

msanatan, 2020-05-04:
Hey @noopkat, thanks for checking it out!

I didn't want to submit a PR without getting a fix for the other issues as well. Hopefully I can put my brain to better use and help out there 😅. If I'm not able to code in a timely manner, I'm definitely happy to test!

And this library is so cool. I love making games for my Arduboy and playing other people's games as well. There wasn't a really easy to upload games to Macs without compiling until I found this repo. Saves me a ton of time with node-serial so I'm grateful haha.

Fingers crossed for the next release, I hope it works out!

noopkat, 2020-05-17:
@msanatan thank you so much for your kind words about this project, despite it being a little bit buggy for you 😅 

I would love any PRs in general even if they're not related to this problem. I just moved house last week so I'm still trying to get back on top of my open source work 📦 

I should be able to get to this issue in particular pretty soon 🤞

msanatan, 2020-05-21:
Sounds like a busy time, good luck with everything!!!

noopkat, 2020-05-24:
Tasks for me to work through in order to start fixing this:

+ [ ] attempt to repro this on Raspbian and Windows first
+ [ ] consider adding a new board property to optionally close (try to avoid this)
+ [ ] follow up with serialport maintainers and possibly help them investigate / file a bug

As a reminder to myself, this is affecting 32u4 emulated USB connections only.