noopkat/avrgirl-arduino #197: Problem with webserial demo example

dvalladaresv, 2020-04-29:
Hello,
First of all, congratulations on the project!

I write to consult a thing that happens to me in the example of demo of webserial (tests/demos/webserial/. I am serving it on a server and the problem it gives me is that sometimes it works perfectly, but other times it gives me an error on line 13102 ==> navigator.serial.requestPort(this.requestOpstions). I am using /dist/avrgirl-arduino.global.js.

I have tried on linux, windows and MacOs, and it happens the same in all of them, which sometimes gets perfectly. But others give permission failure
The error it gives me is the following:

[image]

I understand that the error may be due to you not having permission to open the port.
Do you know of anything I could do to fix the problem?

Muchas gracias por la ayuda y perdonar mi inexperiencia pero estoy aprendiendo sobre webserial.

techwiznixon, 2020-05-18:
Looks like you haven't enabled the access in your browser to the core Chrome features to drive this functionality. This demo runs on the Web Serial API which is currently still experimental, so you'll have to enable it in Chrome.

As stated in the README for this demo, please go to `chrome://flags` and enable the `#enable-experimental-web-platform-features` option. This should fix things for you.

closed 2020-05-19 by dvalladaresv

noopkat, 2020-05-24:
@techwiznixon thanks for the assistance on this issue here I really appreciate it. Feel free to reopen this issue @dvalladaresv if you still can't get it working! 🙇