noopkat/avrgirl-arduino #199: After running firmata-party yun --debug it cant find a file under /junk/hex

wispyco, 2020-05-05:
## Description

After running `firmata-party yun --debug` it cant find a file under this path for avrgirl-arduino node module
```
/home/anders/.nvm/versions/node/v13.6.0/lib/node_modules/firmata-party/node_modules/avrgirl-arduino/junk/hex/yun
```
### Expected behaviour

to flash my arduino yun

### Actual behaviour

```
[Error: ENOENT: no such file or directory, scandir '/home/anders/.nvm/versions/node/v13.6.0/lib/node_modules/firmata-party/node_modules/avrgirl-arduino/junk/hex/yun'] {
  errno: -2,
  code: 'ENOENT',
  syscall: 'scandir',
  path: '/home/anders/.nvm/versions/node/v13.6.0/lib/node_modules/firmata-party/node_modules/avrgirl-arduino/junk/hex/yun'
}
```

## Operating system and version

### Linux and Unix based systems

5.3.0-51-generic #44-Ubuntu SMP Wed Apr 22 21:09:44 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

_For information on the specific distribution. Please consult your distribution documentation._

## Avrgirl Arduino version

Arduino Yun 

## NodeJS version

node v13.6.0
## Arduino Board being used

Arduino yun

I did notice in this repo in the root `/junk/hex there is no arduino yun folder` seen here https://github.com/noopkat/avrgirl-arduino/tree/master/junk/hex

noopkat, 2020-05-17:
hi there @wispyco thanks for reporting - total overlook here on my part. I added a preliminary compiled version of StandardFirmataEthernet to this library and have published a new version if you want to try it out to see if it works with your Yun. 

I don't have a Yun so I can't currently test it. Fair warning I used StandardFirmataEthernet because it seems that this is the compatible version of Firmata for the Yun. Do you have anymore details about this / can confirm on your end at all? I'm not super knowledgable about the Yun unfortunately.

Firmata Party has been updated to use the latest version of Avrgirl Arduino the this Yun patch. Try reinstalling it and let me know if anything works as expected?

Thanks! 🙇

wispyco, 2020-05-17:
I'll give it a try now. I had to use StandatdFirmatePlus to get it to work on my end, but I will test and see if it works on StandardFirmataEthernet and report back. Thanks!

noopkat, 2021-01-10:
closing due to staleness of issue. Feel free to reopen if still broken. Thanks!

closed 2021-01-10 by noopkat