noopkat/avrgirl-arduino #201: [Feature Request] Allow reading and writing to EEPROM

Luberry, 2020-05-25:
I am in the process of writing a configuration program for my [LED Spy](https://gitlab.com/Luberry/smashboxledspy) project, and need to be able to write configuration to EEPROM. I have looked at one of your older projects which is now unmaintained `avrgirl-stk500v2` which had this functionality, but have not been able to get it to work, since this project seems to be maintained, I would like to request the feature here. 

I would add it myself, but javascript is not my forte.

noopkat, 2020-06-21:
Hi @Luberry sorry for the delay in responding I've actually been investigating this for you for the last couple of weeks. A couple of questions if you'd be kind enough to answer (it's ok if you found an alternative in the meantime I'm just curious):

1. Which microcontroller are you using (with the eeprom you want to write to)?
2. Which programmer are you using? Such as an AVRISP mkii or maybe even this is an Arduino board or something like that.

The reason for question 2 is I also am interested in whether your USB interface is serial or libusb. I have fixed stk500v2 to work with serial interfaces and will publish a fix for it soon, but also want to know if libusb is broken on the library as well if that is what you are trying to use.

Anyway your project is cool af and would love to know more details about it and if any of my software can help further beyond this example!

Luberry, 2020-06-21:
I am using serial communications, currently I am just invoking avrdude using execFile, but can easily drop your library in. I am using the Atmega 2560 and is Arduino based, which is the Stk500v2 programmer iirc. I would definitely love to use a library instead of packing binaries in my app. Thanks for the response, I've also open sourced my designer software now that I have an alpha release out [here](https://gitlab.com/Luberry/led-spy-designer). Still a major WIP but at least it's functional now.

noopkat, 2020-07-26:
Hi @Luberry just getting back to you on this. Your project looks super cool! And I totally see why you'd want to stop packing binaries. So I have a [fresh version (2.0.0)](https://github.com/noopkat/avrgirl-stk500v2/releases/tag/v2.0.0) of avrgirl-stk500v2 published with an [updated example](https://github.com/noopkat/avrgirl-stk500v2/blob/master/README.md#example-using-serialport) in the README which I hope helps fix any issues you might have run into. I'd recommend trying it out for the eeprom use case you outlined instead of avrgirl-arduino because it's gonna take me much more work to get eeprom functionality working here.

Let me know how you go; I'd be thrilled to hear if you are able to kick that binary and use a lightweight node lib such as avrgirl-stk500v2 as this is the use case I wrote it for! 😄

Luberry, 2020-09-18:
@noopkat Using a combination of this library for the firmware flash and the 2.0.0 release avrgirl-stk500v2 I have successfully gutted avrdude. For some reason firmware flashes from the stk500 lib would always be corrupt and take forever, but they work with this library so its fine.