noopkat/avrgirl-arduino #202: Flashing stalls at "connected"

rubendax, 2020-05-28:
## Description

I'm trying to flash a 3rd party leonardo clone (Based on ATMEGA 32u4) but it seems to always stall out at the point in which the console logs: "connected". I've tried letting it sit for quite some time but it does not seem to ever progress beyond that point. I have set manual reset to TRUE, because I found that when it was omitted or if it was set to FALSE I would receive the error "could not open board on port XX".

I have used the list function to confirm that it is using the correct product id of 0x8036. I have also tried manually specifying the port name. It seems to be the same result either way.

Thanks so much for your work on this project! Really hoping I can get it to work.

## avrgirl.list()
{ path: '/dev/tty.usbmodem1463201',
    manufacturer: 'Arduino LLC',
    serialNumber: undefined,
    pnpId: undefined,
    locationId: '14632000',
    vendorId: '2341',
    productId: '8036',
    _standardPid: '8036' } 

## Console Log Output:
$ node avrgirl.js
found leonardo on port /dev/tty.usbmodem1463201
reset complete.
connected

## My JS code:

var Avrgirl = require('avrgirl-arduino');

var avrgirl = new Avrgirl({
  board: 'leonardo',
  debug: true,
  manualReset: true
});

var hex = 'MYHEX.hex';

avrgirl.flash(hex, function(error) {
  if (error) {
    console.error(error);
  } else {
    console.info('done.');
  }
});

## Arduino Board being used

It's a [3rd party leonardo / micro clone](https://www.amazon.com/gp/product/B01HCXMBOU) based on the ATMEGA 32u4. Out of the box it is seen as 'micro', but I can change the board name to leonardo (as recommended by the manufacturer) through flashing any sketch in the Arduino IDE. I have a handful of these boards so I have tried using the board as micro and as leonardo, both seem to yield the same result. I have no problem flashing these boards through the Arduino IDE or Platformio. I am hoping to be able to get avrgirl to work with any 32u4 based board as long as I have control or at least visibility of the board name, PID, and VID.

## Step by step guide to reproducing the issue

1. Run **node** command on .js file.
     1. Console outputs the message I copied above ending with "connected".
     2. RX LED on hardware blinks.
     3. Console remains at "connected" indefinitely.

## Operating system and version

Mac OS 10.14.6

## Avrgirl Arduino version

avrgirl-arduino@4.2.3

## NodeJS version

12.17.0

## Text editor and shell

Visual Studio Code

noopkat, 2020-11-01:
Hey @rubendax, sorry for not looking this sooner but I am unable to reproduce using a 32u4 board with node 12.x and this version of Avrgirl on osx (albeit Catalina).

You might have already moved on given the time that has passed, however if you're still looking to use this library and it's still not working on the latest version, please reopen this issue and we can debug it together.

closed 2020-11-01 by noopkat

rubendax, 2020-11-01:
Hi @noopkat thanks for getting back to me. Unfortunately I am still experiencing the same issue. Peculiar that it works for you. I wonder what variables there are that I have not considered. I've tried with three different 32u4-based dev boards, every time the process stalls at "connected".

rubendax, 2020-11-22:
#### Fixed
I was able to get past this problem. Sorry for any wasted time. 
I might be totally wrong, but I think my fix was actually updating to Mac OS **10.15 Catalina**... 
I'm not sure if others would be able to reproduce this, I'm definitely a bit of a newb so it is very possible that I was doing something wrong with NodeJS, especially back in May I first opened this issue. I've now tested with node versions 10, 12, and 15 and they are all now working with avrgirl-arduino and 32u4 based boards in Mac OS Catalina.
I have also tested in Mac OS **11 Big Sur** (on a different computer) and it is also working there.

##### Additional info in case it's interesting:
I started trying to build avrgirl-arduino into the backend of simple GUI electron app, and I was running into strange errors where the app would not function while Chrome (or any chromium based browser) was running in the background. I tested on two different computers which were both running Mac OS **10.14 Mojave**. As I had never used electron before, I figured it might be some insurmountable conflict between electron's embedded chromium and the web browser. Fast forward a few days and Big Sur was released, meaning I finally felt comfortable upgrading to Catalina 😅. 
Miraculously, all of my problems (avrgirl & chrome conflict) were fixed after the update.

Thanks so much for all of your work on this library!!
My next goal is to hopefully help solve the Chrome web-serial incompatibility with 32u4 based boards.

noopkat, 2020-11-22:
Hi @rubendax I am SO GLAD that the upgrade fixed your issue!! I have been really scratching my head about this and just released a more in depth debugging feature to try to figure out why you were still experiencing the issue 😅 

Thanks for taking the time to detail all of the things you tried and what seemed to fix things. It's VERY helpful for me should we have another developer run into the same thing on Mojave 🙌 

Also, if you can share or have the time to respond, I'm curious about what you're building with avrgirl! I love hearing about Electron apps that flash boards, so feel free to share with me if you like 😄 

I'm really excited you're unblocked and thanks for your kind words.

[image]

rubendax, 2020-11-22:
😊
I am also so glad that I was able to get past this issue! SO much better to use an active open-source JS library like avrgirl-arduino than trying to embed avrdude 0_0
Sorry you spent so much time and built a whole new debugging feature for my issue...

Sure @noopkat I'm happy to share. I'm a musician and I design and build my own music tech hardware - specifically MIDI controllers - which are based on AVR MCUs and are programmed using the arduino framework. For the past year I've been in the process of transforming my hobby into a solo business and so I'm trying to create a reliable system for my future hardware customers to be able to update the firmware of their devices as I add new features, fix bugs, etc. 
So I'm using avrgirl-arduino to create a firmware deployment & updating system for my devices.

Thus far I've created a basic electron app to achieve this - but ideally I would love to get it to run in the browser. I am already building a web-app which interfaces with the hardware devices for the user to customize their device settings/parameters, so it would be amazing if firmware-updating could be a feature hosted at the same location. Going to take a look at 
**/lib/connection-browser.js** to see if I can hack some 32u4 functionality into it - though this might be beyond my current ability
¯\\__(ツ)_/¯

Separate, but related, I may have a potential contribution to make to avrgirl-arduino (but I need to do a bit more testing first and need to make sure I'm following the proper contribution protocol):
I discovered that my avrgirl-arduino electron app works great if the device is connected over a standard serial connection, however, on Mac, opening the port would fail if the device to which you are trying to upload is currently connected with a **native-USB HID or MIDI** connection. I've hacked in a fix in **/lib/connection.js** which solves this problem by opening a new temporary serial port in the middle of the flashing process. 
I'm not sure if anyone else has experienced this problem, but native USB is one of the main reasons people choose to use ATMEGA32u4, so I figure it might be useful for others.
I'm still a git baby 👶 but I'm happy to try to open a proper issue/pull-request for it.

Thanks again for everything!

noopkat, 2020-12-06:
@rubendax your project sounds super cool thank you for sharing! 😄 

I think that your use case is a perfect one for using avrgirl - it's totally why I wrote the library in the first place–to solve this problem for myself 😅 

I really would love to see 32u4 working in a nice flow within the browser. It's something I have failed to make the time for this year but that doesn't mean it's not important and we really should have it as an option.

Please watch this space in case I do get it up and running. Thank you so much again for sharing your use case as it helps keep me motivated to improve the existing browser support for this library 💯  Let me know if you do end up spelunking around in the code and hack something together 😄