noopkat/avrgirl-arduino #240: React Demo build error with module imports

mleanos, 2021-08-30:
# Issue Template

Please review the [contribution](https://github.com/noopkat/avrgirl-arduino/blob/master/CONTRIBUTING.md) and [code of conduct](https://github.com/noopkat/avrgirl-arduino/blob/master/CODE_OF_CONDUCT.md) guidelines.

## Description

_If you raise an issue on bug/issue please provide the requested information, where not possible, please provide as many details as possible and ultimately a thorough description of your issue._

### Expected behaviour

_What behaviour did you expect did occur?_
After a clean install of the [React Demo](https://github.com/noopkat/avrgirl-arduino/tree/master/tests/demos/webserial/react-demo), I get an error when running `npm start` should build successfully.

### Actual behaviour

_What unexpected behaviour occurred?_
Build produces the error: `./src/style.css (./node_modules/css-loader/dist/cjs.js??ref--6-oneOf-3-1!./node_modules/postcss-loader/src??postcss!./src/style.css)
Syntax error: Unknown word (1:1)`

## Operating system and version

_Please provide version on the operating system etc. used._

### Windows (pre version 7)

`C:\> ver`

### Windows (Version 7 and newer)

`C:\> systeminfo | findstr /B /C:"OS Name" /C:"OS Version"`
OS Name: Microsoft Windows 10 Professional
OS Version: 10.0.19043 Build 19043

### Linux and Unix based systems

`$ uname -a`

_For information on the specific distribution. Please consult your distribution documentation._

## Avrgirl Arduino version

_Please provide version on the Avrgirl Arduino version used_

_If working in a fork of the GitHub project, obtain this information post installation using:_

`$ npm list | grep avrgirl-arduino`

_If you are working on code, which has diverged from the lastest release, please mention this_.

## NodeJS version

_Please provide version on the Node version used._

`$ node --version`
`v12.18.1`

## Arduino Board being used

_Please provide version/model on the Arduino board being used._

_A full List of boards and processor names are [available on Wikipedia](https://en.wikipedia.org/wiki/List_of_Arduino_boards_and_compatible_systems)._

## Log output, if available

_Please add any log output, which can be used to describe the issue or circumstances surrounding the issues._ 

There is a property you can use named `megaDebug` that is passed into the instance options. This will output much more low level debugging info that could help the author debug your issue more quickly. See below for how to use it:

```javascript
var avrgirl = new Avrgirl({
  board: 'uno',
  debug: true,
  // turn on megaDebug mode!
  megaDebug: true
});
```

## Step by step guide to reproducing the issue

_If possible a code snipppet to demonstrate the issue or a throught step by step description of how to get reproduce the error._
Running the example based on the steps described in the [React Demo README](https://github.com/noopkat/avrgirl-arduino/tree/master/tests/demos/webserial/react-demo#how-to-run-this-example).

If I comment out the [CSS import here](https://github.com/noopkat/avrgirl-arduino/blob/master/tests/demos/webserial/react-demo/src/App.js#L3), I get another error which seems to be a symptom of the same issue: 

```
Failed to compile.

./src/avrgirl-arduino.js
  Line 1:1:  Parsing error: Unexpected token

> 1 | ../../../../../dist/avrgirl-arduino.js
    | ^
```

I'm not familiar with the technique of importing files/modules that these two files are attempting to accomplish. Perhaps this is a limitation of being on a Windows machine: [CSS import file](https://github.com/noopkat/avrgirl-arduino/blob/master/tests/demos/webserial/react-demo/src/style.css) & [avrgirl-arduino module](https://github.com/noopkat/avrgirl-arduino/blob/master/tests/demos/webserial/react-demo/src/avrgirl-arduino.js)

kidfromjupiter, 2024-08-15:
Ran into this as well. How did you fix this?

kidfromjupiter, 2024-08-15:
I think this may be due to some error in outdated dependencies. I copied the App.js into a new Nextjs project and patched 2 lines in avrgirl-arduino.js. Have no idea why it didn't work or why my patch worked. But it works with NextJS and uploads successfully to my Uno.