noopkat/avrgirl-arduino #241: SAM-D2x Support

ma-ku, 2021-09-09:
## Description

I am currently trying to find a way to flash an Arduino MKR 1010 Wifi using avrgirl-arduino. After tinkering around with different board configurations i realized that none of the supported three protocols seem to suit the protocol expected by the bootloader which seems to be SAM-BA. Now my question if this can be solved with avrgirl-arduino as-is or if we need to add SAM-BA as another protocol?
 
### Expected behaviour

Arduino MKR* boards can be flashed with this library
 
### Actual behaviour

Could not find a board configuration that would allow flashing above chips.

## Arduino Board being used

Arduino MKR 1010 WIFI

rubendax, 2021-11-13:
It would be amazing to have SAMD support in avrgirl-arduino, but as this is a library made specifically for **AVR** devices, as opposed to **ARM**, it seems like this request is probably outside of the scope...

That being said, if any contributors are looking to expand this project to support SAMD chips, [BOSSA](https://github.com/shumatech/BOSSA) is an open source, lighter, and more compatible option that could be used instead of the SAM-BA software. This is what Adafruit uses to upload to their SAMD-based boards. More BOSSA info can be found [here](https://www.shumatech.com/web/products/bossa).

ma-ku, 2021-11-13:
Actually I have partly ported bossa to typescript using web-midi. So   ##flashing and many other functions of the library are already there. Feel free to play with it. [bossa-web](https://github.com/ma-ku/bossa-web)

rubendax, 2021-11-14:
@ma-ku Whoa, amazing! I love that you are porting bossa to JS.
I would be very happy to help test or anything like that if needed. I'm far from a JS expert but I've used avrgirl-arduino pretty extensively.

ma-ku, 2021-11-14:
Well, it’s quite complete already. Some flashers are still missing but porting is straightforward. I use this code for flashing MKR 1010 WIFIs from a web application